openscad/openscad #5595: expose a variable that contains the name of the current preset

krhackbarth, 2025-01-18:
**Is your feature request related to a problem? Please describe.**
I want to do different processing when the current preset is "design default values" than when the value is anything else.  To do that I need to be able to determine that value in my code.

**Describe the solution you'd like**
Provide a standard variable ($preset or something more appropriate) that contains the name of the currently selected preset.

**Describe alternatives you've considered**
I can't think of an alternative.

**Additional context**
None

t-paul, 2025-01-18:
Can you show an example why that's needed? It seems a bit dangerous to make this a 2-way street for information that's not explicitly defined in the customizer in a way that's visible to the user.

krhackbarth, 2025-01-19:
I'd be very happy with a read-only variable.  

I use OpenSCAD to create a customizer for keyguards.  I don't expect you to know what those are, but I can say that my designer supports hundreds of independent customization parameters.  The default values are just placeholders.  I've gone through several upgrades of the designer.  In the last couple of releases, I've introduced new parameters that substitute for existing ones.  My users will be upgrading with 10's of existing Presets in their JSON files.  I've tried to provide feedback to these users (generally non-technical) as to how to assign values to these new parameters based on the old (and now unused) parameters.  I do this by echoing instructions to the Console pane.  I want to bypass these recommendations and the processing to produce them if the Preset is the default one (which loads upon start-up) so my users don't think they need to change the default parameter values.

t-paul, 2025-01-19:
Right, so the minimal thing that would cover the case would be a function doing `customizer_is_default_set_selected()` -> `true/false`. This should also work fine then running via command line.

To make it easier to extend later, it's probably better using a more general approach, like `customizer_status()` returning a data structure that can have more fields added later.

krhackbarth, 2025-01-19:
I think either approach would work for me.  I'm not as adept at processing data structures as I am at testing true/false.