openscad/openscad #5599: Zooming on MacOS causes crash

Johnr24, 2025-01-19:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Open Project in Openscad, 
2. Zoom in Window 

**Expected behavior**
openscad to not crash on zooming out on the vlewport

**Environment and Version info (please complete the following information):**
 - OS: MacOS 15.2
 - System: Macbook Pro, 13-inch, M1, 2020
 - OpenSCAD Version, occuring on nightly, tested as far back as version,  2024.12.20 and issue was still happening, Issue is not occuring on 2024.10.28

kintel, 2025-01-19:
I cannot reproduce here neither on macOS 14 or 15. Could you provide a crash log from Console.app, or some more info about how you reproduce this?

Johnr24, 2025-01-19:
sure here is the crash log 

 Translated Report (Full Report Below)
-------------------------------------

Process:               OpenSCAD [34369]
Path:                  /Applications/OpenSCAD.app/Contents/MacOS/OpenSCAD
Identifier:            org.openscad.OpenSCAD
Version:               2025.1 (2025.01.12)
Code Type:             ARM-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2025-01-19 15:08:16.4467 +0000
OS Version:            macOS 15.2 (24C101)
Report Version:        12
Anonymous UUID:        4ECAEEEA-0E6E-02EA-BCD5-24747D9FED9B

Sleep/Wake UUID:       977A253D-1130-414D-8B39-7A288AC84B3C

Time Awake Since Boot: 400000 seconds
Time Since Wake:       2868 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000000
Exception Codes:       0x0000000000000001, 0x0000000000000000

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [34369]

VM Region Info: 0 is not in any region.  Bytes before following region: 4362928128
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      __TEXT                      1040d0000-104ae0000    [ 10.1M] r-x/r-x SM=COW  /Applications/OpenSCAD.app/Contents/MacOS/OpenSCAD

Thread 0 Crashed::  Dispatch queue: com.apple.main-thread
0   QtWidgets                     	       0x1079b4bf0 QWidget::hide() + 0
1   QtWidgets                     	       0x107ad4b50 QMainWindowLayout::applyState(QMainWindowLayoutState&, bool) + 1296
2   QtWidgets                     	       0x107ad4620 QMainWindowLayout::setGeometry(QRect const&) + 792
3   QtWidgets                     	       0x107999284 QLayoutPrivate::doResize() + 344
4   QtWidgets                     	       0x1079994d0 QLayout::activate() + 408
5   QtWidgets                     	       0x10796cc2c QApplicationPrivate::notify_helper(QObject*, QEvent*) + 228
6   QtWidgets                     	       0x10796e60c QApplication::notify(QObject*, QEvent*) + 3308
7   OpenSCAD                      	       0x10472450c OpenSCADApp::notify(QObject*, QEvent*) + 28
8   QtCore                        	       0x108db524c QCoreApplication::notifyInternal2(QObject*, QEvent*) + 292
9   QtCore                        	       0x108db615c QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*) + 492
10  libqcocoa.dylib               	       0x10802ea78 QCocoaEventDispatcherPrivate::processPostedEvents() + 296
11  libqcocoa.dylib               	       0x10802f5fc QCocoaEventDispatcherPrivate::postedEventsSourceCallback(void*) + 560
12  CoreFoundation                	       0x183c2c894 __CFRUNLOOP_IS_CALLING_OUT_TO_A_SOURCE0_PERFORM_FUNCTION__ + 28
13  CoreFoundation                	       0x183c2c828 __CFRunLoopDoSource0 + 176
14  CoreFoundation                	       0x183c2c58c __CFRunLoopDoSources0 + 244
15  CoreFoundation                	       0x183c2b128 __CFRunLoopRun + 840
16  CoreFoundation                	       0x183c2a724 CFRunLoopRunSpecific + 588
17  HIToolbox                     	       0x18f182530 RunCurrentEventLoopInMode + 292
18  HIToolbox                     	       0x18f18817c ReceiveNextEventCommon + 216
19  HIToolbox                     	       0x18f188508 _BlockUntilNextEventMatchingListInModeWithFilter + 76
20  AppKit                        	       0x187795034 _DPSNextEvent + 660
21  AppKit                        	       0x1880f92d4 -[NSApplication(NSEventRouting) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 688
22  AppKit                        	       0x187788060 -[NSApplication run] + 480
23  libqcocoa.dylib               	       0x10802de9c QCocoaEventDispatcher::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) + 1904
24  QtCore                        	       0x108dbebc4 QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) + 532
25  QtCore                        	       0x108db58d8 QCoreApplication::exec() + 112
26  OpenSCAD                      	       0x1046d1fb8 gui(std::__1::vector<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>, std::__1::allocator<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>>>&, std::__1::__fs::filesystem::path const&, int, char**) + 2192
27  OpenSCAD                      	       0x1043b5be4 main + 18796
28  dyld                          	       0x1837c4274 start + 2840
[... 1036 lines cut ...]
      "factorPackIds" : {

      },
      "deploymentId" : 240000012
    }
  ],
  "experiments" : [

  ]
}
}

Model: MacBookPro17,1, BootROM 11881.61.3, proc 8:4:4 processors, 8 GB, SMC 
Graphics: Apple M1, Apple M1, Built-In
Display: Color LCD, 2560 x 1600 Retina, Main, MirrorOff, Online
Memory Module: LPDDR4, Hynix
AirPort: spairport_wireless_card_type_wifi (0x14E4, 0x4378), wl0: Sep 28 2024 04:40:32 version 18.20.462.0.7.8.175 FWID 01-10fa77df
IO80211_driverkit-1345.8 "IO80211_driverkit-1345.8" Nov  9 2024 17:02:32
AirPort: 
Bluetooth: Version (null), 0 services, 0 devices, 0 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
USB Device: USB31Bus
USB Device: USB31Bus
Thunderbolt Bus: MacBook Pro, Apple Inc.
Thunderbolt Bus: MacBook Pro, Apple Inc.

Johnr24, 2025-01-19:
I just moved that plist out of the directory and sadly it's not fixed it. the second I touch the pan window it crashes out.

kintel, 2025-01-19:
Before digging further, try kill the plist cache after removing:

1. Make sure OpenSCAD is _not_ running
2. In a terminal, run `rm ~/Library/Preferences/org.openscad.OpenSCAD.plist; killall -u $USER cfprefsd`

kintel, 2025-01-19:
Btw., are you using multiple screens?

Johnr24, 2025-01-19:
thank you, that in the above has fixed

closed 2025-01-19 by Johnr24

kintel, 2025-01-19:
Thanks for verifying! We still need to fix this, as it's not something we can expect users to figure out. 
We'll continue this in #5486, which is a duplicate of this issue.