openscad/openscad #5601: Hull() with grouping

Harvie, 2025-01-19:
I think it would be really beneficial for designing tubes, manifolds and similar structures, if hull() had options to group N items (eg. 2 for pairs) instead of having to hull them independently.

```openscad
//This
hull(group=2) { A; B; C; D; }

//Would be same as
union() {
  hull() { A; B; }
  hull() { B; C; }
  hull() { C; D; }
  //hull() { D; A; } //Only for hull(group=2, wrap=true) {...
}
```

[image]

UBaer21, 2025-01-19:
```
module CHull(end=true){
  for(i=[0:$children - (end?2:1)])hull(){
    children(i);
    children( (i+1)%$children);
  }
}
```

Harvie, 2025-01-20:
@UBaer21 i know this can be done... Same way you can write your own `intersection_for()` using `intersection()` and `for()`. That doesn't mean it's not useful to have it natively implemented in OpenSCAD.

UBaer21, 2025-01-20:
No you can't
```
intersection(){
  for(i=[0,120,240])rotate(i)translate([5,0])sphere(10);
}
```
Will only work with lazy union, you need to write this recursively to intersect the intersection with every object.
Something like a hull_for() would be more useful but require to have special variables to use with the children so it will become more complex than using a chained hull. 
Making a chained hull without a loop by using multiple objects is quite rare. I made that module a while ago and haven't used it since.
You sure want "group" to be the number of children if not defined else it would make the use of hull problematic - or did you want a new module for this function?
Design a manifold wouldn't be so much better either as you need another hull for every fork.  And for a tube you need a difference too.
So i am not convinced at the moment it is worth to complicate hull() for that if you can use a simple loop module.