openscad/openscad #5605: Poor performance on command line invocation

MethylBromide, 2025-01-20:
**Describe the bug**
I observed this issue while working with this model: [https://www.thingiverse.com/thing:1646307](https://www.thingiverse.com/thing:1646307)
The package includes a MS Windows "BAT" file that repeatedly calls OpenSCAD with different command line parameters to generate a set of STL files. While it was running, I observed it seemed to take very long for each one. Looking at the timestamps on the resulting files, I observed the timing averaged 3m5s per model generated.

As a test, I then ran the same script manually in the OpenSCAD UI for just one set of parameters. This took 17 seconds, starting from my UI action to launch the program, until I had completed all steps and closed the program. I can do the manual operation in 1/10th the time it takes to execute the automated version.

So it appears there's a substantial time penalty for using the command-line option -- I know the Windows BAT file execution isn't adding much to this. A similar BAT file to merely echo the arguments runs in less than a second.

**To Reproduce**
Steps to reproduce the behavior:
1. Use a Windows 11 computer with 64-bit processor.
2. Download the above-linked ZIP package from Thingiverse and unzip it somewhere.
3. Follow the included instructions to create directories to contain the generated STL files.
4. In File Manager, double-click the generate.bat file to execute the script.
5. Wait. Depending on the speed of your computer it may take some hours.
6. Observe the timestamps on the resulting files. The files in each subfolder (e.g. .../STL/Metric/Large) were generated consecutively, so you can calculate the average time via (oldestTimestamp - newestTimestamp)/(n-1) where n is the number of files. In my case this came out to 3m5s.
7. Prepare a stopwatch. Close OpenSCAD if it is open. From File Explorer, launch RadiusGauge-MM.scad and simultaneously start timing. Press F6, then F7, accept the default STL filename and save. Exit OpenSCAD. How long did that take?

**Expected behavior**
I would expect the command-line script to take much less time than doing the same thing manually. The command line option has less work to do, after all -- it doesn't have to mange the UI, generate a preview, and so on.

**Code reproducing the issue**
See above link.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment and Version info (please complete the following information):**
 - OS: Windows 11 64-bit
 - System: Intel PC 64-bit
 - OpenSCAD Version: 2025.01.06 (git 92a13b4f0) from openscad.org website.

**Library & Graphics card information**

**Additional context**
During automated execution, I observed the following message on the Windows "Command" window: "Fontconfig error: Cannot load default config file: No such file: (null)". This has been reported previously in #4919 . I don't know whether this is related.

SyntaxBlitz, 2025-01-20:
The commandline defaults to the (older, slower) CGAL backend; it sounds like you might be using the (newer, faster) Manifold backend from within the GUI (not sure whether this is the default yet, but you can check in `Edit > Preferences > Advanced > 3D Rendering > Backend`, assuming your build is from before #5589).

You can use the Manifold backend from the CLI with `--backend=manifold` as an argument. On the file you linked, that brings total CLI wall time on my machine from 12.1s to 0.35s.

t-paul, 2025-01-26:
Closing, assuming the issue was the missing "manifold" selection described above.

closed 2025-01-26 by t-paul