openscad/openscad #5607: Measuring tools stopped working in recent snapshots.

Nikorasu, 2025-01-21:
After recently switching back to Windows for a bit, I just happened noticed that, in at least the last few snapshots I've downloaded, the Measuring tools that normally work after rendering, no longer function. I use them a lot and find them essential for ensuring part accuracy.
It's currently still an issue with the 2025.01.20-x86-64 version although I'm not sure where it began. I'm on Windows 10 if that matters.

t-paul, 2025-01-21:
You need to enable the Manifold back-end.

https://fosstodon.org/@OpenSCAD/113256867413539398

closed 2025-01-21 by t-paul

Nikorasu, 2025-01-21:
> You need to enable the Manifold back-end.
> 
> https://fosstodon.org/@OpenSCAD/113256867413539398

Oh woops, sorry.. I totally missed/forgot about that.. Thanks!