openscad/openscad #5610: Random number generation gives similar results on calls with the same seed

Zecyl, 2025-01-21:
**Describe the bug**
When generating random numbers using `rands()`, calls with the same seed, but with different `min_value`, `max_value`, and `count` values generate similar numbers.

**To Reproduce**
Steps to reproduce the behavior:
1. Create new .scad file
2. Copy paste the minimal example provided
3. Preview model
4. Similar values for a and b get generated

**Expected behavior**
Different calls to rands should yield different values.

**Code reproducing the issue**
``` openscad
for (i = [1 : 5]) {
    seed = rands(0, 1000000, 1)[0];
    a = rands(0, 50, 1, seed)[0];
    b = rands(1, 49, 1, seed)[0];
    echo(a);
    echo(b);
}
```
``` openscad
seed = rands(0, 1000000, 1)[0];
a = rands(0, 50, 3, seed);
b = rands(1, 49, 5, seed);
echo(a);
echo(b);
// The first three values of a and b are similar
```

**Screenshots**
[image]

**Additional context**
I am currently trying to make my own random crystal cluster 3D model generator, using OpenSCAD. (I am aware of the other crystal 3D model generators out there)

**Environment and Version info :**
 - OS: Windows 11
 - System: Dell PC 64-bit
 - Official OpenSCAD Version 2021.01

**Library & Graphics card information**
```
OpenSCAD Version: 2021.01
System information: Windows(TM) 10.0 SP 0.0 NTW 1 MSDN 724833 12 CPUs 15.69 GB RAM
User Agent: OpenSCAD/2021.01 (Windows(TM) 10.0 SP 0.0 NTW 1 MSDN 724833)
Compiler: GCC "7.4.0" 64bit
MinGW build: MingW64
Debug build: No
Boost version: 1_60
Eigen version: 3.3.7
CGAL version, kernels: 4.11, Cartesian<Gmpq>, Extended_cartesian<Gmpq>, Epeck
OpenCSG version: OpenCSG 1.4.2
Qt version: 5.13.0
QScintilla version: 2.11.2
InputDrivers: 
GLib version: 2.50.2
lodepng version: 20180910
libzip version: 1.5.2
fontconfig version: 2.13.1
freetype version: 2.10.1
harfbuzz version: 2.4.0
cairo version: 1.16.0
lib3mf version: 1.8.0
Application Path: C:/Users/ulyss/Downloads/OpenSCAD
Documents Path: C:\Users\ulyss\OneDrive\Documents
User Documents Path: C:\Users\ulyss\OneDrive\Documents
Resource Path: C:/Users/ulyss/Downloads/OpenSCAD
[... 245 lines cut ...]
GL_KHR_debug
GL_KHR_no_error
GL_KHR_shader_subgroup
GL_KHR_shader_subgroup_arithmetic
GL_KHR_shader_subgroup_ballot
GL_KHR_shader_subgroup_basic
GL_KHR_shader_subgroup_clustered
GL_KHR_shader_subgroup_quad
GL_KHR_shader_subgroup_shuffle
GL_KHR_shader_subgroup_shuffle_relative
GL_KHR_shader_subgroup_vote
GL_KHR_texture_compression_astc_ldr
GL_NV_blend_square
GL_NV_conditional_render
GL_NV_primitive_restart
GL_NV_texgen_reflection
GL_NV_timeline_semaphore
GL_OVR_multiview
GL_SGIS_generate_mipmap
GL_SGIS_texture_edge_clamp
GL_SGIS_texture_lod
GL_SUN_multi_draw_arrays
GL_WIN_swap_hint
WGL_EXT_swap_control
```

kintel, 2025-01-21:
You're providing the same seed for the two runs, so the underlying rng generator will generate the same data. If your range is similar, clamping to your range may result in very similar results.
We simply clamp data to a range using https://en.cppreference.com/w/cpp/numeric/random/uniform_real_distribution.

I'm closing this as works-as-intended. If you really want different values, just make ad adjustment to the seed.

Zecyl, 2025-01-21:
Ah I think I understand : so when I pass a seed to the `rands()` function, it just uses it to get a number between the min and max values, but it will always be a number the same "distance" between min and max. So if I just change min and/or max, the value will still be at the same "distance" from both. For example a "translation" in the min and max (adding a certain value) will just give the same result translated by the same amount.

Could you correct me if my understanding is wrong ? Sorry to bother you 😅

kintel, 2025-01-22:
Yeah, that's what it looks like. The underlying C++ library function doesn't promise that this is always the case though, so it may differ from compiler to compiler, so I wouldn't rely on it.

jordanbrown0, 2025-01-22:
If you give the same seed and the same min/max to rands(), you'll get the same value.

Mostly, you don't need to pass a seed at all, unless you need your runs to be repeatable.  If you do need your runs to be repeatable, you should pass a seed to **one** rands() call at the beginning, to "seed" the random number generator.  You would probably discard the result from that first call to rands().

This is all the essence of the behavior of pseudo-random number generators.  You start them with a seed (often defaulting to one supplied by the infrastructure) and they produce a sequence of numbers that don't appear to have anything to do with one another.  Given the same seed, you get the same sequence.  Range limits might affect the values, but only in how they transform the underlying random number into the desired range; they are *not* part of the randomization process.

Net, usually the pattern would look something like so:

```
for (i = [1 : 5]) {
    a = rands(0, 50, 1)[0];
    b = rands(1, 49, 1)[0];
    echo(a);
    echo(b);
}
```

or if you need each successive run to be repeatable:

```
seed = 0;    // Change this to get a different sequence
junk = rands(0, 1, 1, seed);

for (i = [1 : 5]) {
    a = rands(0, 50, 1)[0];
    b = rands(1, 49, 1)[0];
    echo(a);
    echo(b);
}
```

closed 2025-01-22 by jordanbrown0

Zecyl, 2025-01-22:
Ah okay, I understand it now, thank you for your time. And keep doing what you do, OpenSCAD is a wonderful tool ! 👍