openscad/openscad #5614: Segmentation fault with "-o something.echo -o something.stl"

jordanbrown0, 2025-01-23:
**Describe the bug**
CLI with -o something.echo -o something.stl yields segmentation fault

**To Reproduce**
```
$ cat y.scad
cube();
echo("y");

$ master/b/openscad -o y.echo -o y.stl y.scad
Segmentation fault
```

**Expected behavior**
Not a segmentation fault :-)  Based on the usage output "May be used multiple time for different exports", I expected to capture an STL and the echo output.

**Environment and Version info (please complete the following information):**
 - OS: Windows 11 + MSYS2/MINGW
 - System: Intel PC 64-bit
 - OpenSCAD Version: local build (no modifications), ca 09 Dec 2024

**Additional context**
This is also unexpected, not capturing the echo output:
```
$ master/b/openscad -o y.stl -o y.echo y.scad

ECHO: "y"
Geometries in cache: 1
Geometry cache size in bytes: 848
CGAL Polyhedrons in cache: 0
CGAL cache size in bytes: 0
Total rendering time: 0:00:00.000
Top level object is a 3D object (PolySet):
   Convex:       yes
   Facets:         6
```

jordanbrown0, 2026-10-19:
I'll look at this some day.

thalisantunes, 2026-10-31:
/assign-me

closed 2026-10-31 by t-paul