openscad/openscad #5615: New release?

svenstaro, 2025-01-24:
Hey, the last release was 4 year ago. I think it might be a good idea to make a new release as there have been a lot of commits in the mean time. What do you think?

t-paul, 2025-01-24:
See #3640

closed 2025-01-24 by t-paul

alerque, 2025-01-24:
I've looked over the related issue (started 3+ years ago) and understand fully how FOSS development goes when you don't have sponsors. I'll just chime it to say that a release cadence is maybe more important than this project seems to think, and maybe some of the "blockers" are not as big a blocker as they would otherwise be in the context of "the last release can't be used". This project is getting increasingly harder to build an install at the last release point. For example the current Arch Linux packaging no longer builds at all. I was looking into what it would take to get it to build with the newly release boost version, but it doesn't even build with the old boost version right now (possibly gcc updates broke this?). Basically what is happening is by not making periodic releases you're forcing distro maintainers (who make it easy for end users to pickup trying out/using a project) to take on the role of developers and sort through the commit history looking for things to backport to try to keep old versions working. That's largely wasted effort that we'd rather contribute towards keeping fresh things working together.