openscad/openscad #5620: Openscad no longer adds default file suffix when exporting STL

dan-p3rry, 2025-01-25:
**Describe the bug**
In the past when exporting a stl file, OpenSCAD would add the ".stl" to the file name so I don't have to type it.  It doesn't in 2025.1.19 version, and I don't think it did in the version from early January.

**To Reproduce**
Steps to reproduce the behavior:
1. click STL icon in main frame of OpenSCAD GUI
2. Type part name, hit enter

**Expected behavior**
If I type "small_parts" into the Export STL dialog box, file should be saved as small_parts.stl.

**Code reproducing the issue**
If applicable, add trimmed down, self-contained code demonstrating the issue.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment and Version info (please complete the following information):**
 - OS: Win 11
 - System:  Intel PC 64-bit
 - OpenSCAD Version 25.1.19

**Library & Graphics card information**
If applicable, add the content of the Help->Library Info window, or the output of running the `openscad --info` command.

**Additional context**
Add any other context about the problem here.

drpeppershaker, 2025-01-30:
Same issue on Mac as well. It allows you to write over the default extension. Adding extension in finder makes the file readable. Or adding the .stl in the export dialog box works as well. Bit of a pain to remember to do that each time.

MacOS 14.5
Mac M1 Max
OpenSCAD version 2025.01.27

kintel, 2025-02-01:
Could someone explain how to reproduce this? It looks fine here:
* using the OpenSCAD-2025.01.31 snapshot downloaded from openscad.org on macOS 14
* Open OpenSCAD
* type "cube();"
* hit F6
* Chose STL button in the toolbar
* The default filename is "Untitled.stl" where the "Untitled" part is selected
* Type in "my_output" + enter
* -> File is saved as "my_output.stl"

drpeppershaker, 2025-02-01:
I don't know why it's like this for some of us and not others.  "Untitled.stl" is fully highlighted on export using either the STL button in the toolbar or F7. Doesn't matter if it's ascii or bin STL.

[image]

[image]

Interestingly enough on an old dev snapshot that I've been working out of for a while now doesn't have this issue. 
2023.07.17 actually hides the extension all together in the export window, but the file does save with .STL extension.

[image]

drpeppershaker, 2025-02-01:
I don't know if this is relevant or not, but I do have 'Show all filename extensions' disabled in my Finder settings

kintel, 2025-02-01:
> I don't know if this is relevant or not, but I do have 'Show all filename extensions' disabled in my Finder settings

Oh, yes. That does indeed change things. Thanks! I can reproduce it here now.
I don't have a good solution yet; I've never used a computer with hidden file endings; it just feels wrong ;)

drpeppershaker, 2025-02-01:
> Oh, yes. That does indeed change things. Thanks! I can reproduce it here now. I don't have a good solution yet; I've never used a computer with hidden file endings; it just feels wrong ;)

Yeah it's really weird. I didn't even know that was a setting on my mac, because I actually do see my file extensions in finder. I just found the option after doing some googling.

Can also confirm that changing that Finder setting gives me your result--Untitled being highlighted and .stl is not

dan-p3rry, 2025-05-25:
I just noticed that writing out a .scad file works perfectly, I can leave off the .scad in the dialog box and .scad gets appended as file suffix.  Writing out STL, .stl does not get appended.  Just tried .3mf, same problem as .stl files.

(Win 11)

jordanbrown0, 2025-11-23:
I see the same problem on Windows 11, for *.3mf exports.  I say "export 3MF", it pops a save box (with nothing selected), I select the name (the whole name, because that's easy), type a new name (without extension), and it saves it without the extension.

OTOH, when I do a Save-As, it pops a save box, with the whole file name selected, I type a name without an extension, and when I save it adds the extension back on.

stro, 2026-11-12:
Still happens in OpenSCAD 2026.11-RC0 on Windows. The issue was introduced in commit 354d82627e7cd35f2945c5e14500d6704083bf5b. 

I created a pull request for the fix.