openscad/openscad #5632: Non-writeable directory save file issues no error

khklatt, 2025-01-29:
Please first consider that general questions about using OpenSCAD or how to write code for specific models are better asked on the mailing list. See https://openscad.org/community.html for more details.

**Environment**
Which environment is the issue about, e.g. Compiling with MSYS2 on Windows

**Issue**
Explain what the issue is, add code, logs, screenshots as needed
Linux 6.6.67-gentoo-x86_64 #1 SMP PREEMPT_DYNAMIC Mon Jan  6 21:49:27 CST 2025 x86_64 AMD Ryzen 9 6900HX with Radeon Graphics AuthenticAMD GNU/Linux

[OpenSCAD](https://www.openscad.org/) version 2024..1.1. (git bc0d078e0)

Issue - When saving a design to a directory that does not allow write access, no error is reported.

kintel, 2025-01-29:
You're using a very old version, could you retest using a recent build?

khklatt, 2025-01-30:
I use gentoo, and that is the version it will build.  I can download the
source and recompile manually, will give it a try.  Thanks.

Kenny Klatt

khklatt, 2025-01-31:
I tried to compile the latest version of openscad, but received the below
error.  I followed the README.md file.  Installed the KDE Development
packages, no errors.  The check of requirements was all OK.

src/ProgressWidget.cc: In constructor
‘ProgressWidget::ProgressWidget(QWidget*)’:

src/ProgressWidget.cc:11:25: error: ‘class QTime’ has no member named ‘start’

  11 |         this->starttime.start();
     |                         ^~~~~
src/ProgressWidget.cc: In member function ‘int ProgressWidget::elapsedTime()
const’:
src/ProgressWidget.cc:27:32: error: ‘const class QTime’ has no member named
‘elapsed’
  27 |         return this->starttime.elapsed();
     |                                ^~~~~~~
make: *** [Makefile:2737: objects/src/ProgressWidget.o] Error 1

kintel, 2025-01-31:
That's a new error to me. Which version of Qt do you have?

khklatt, 2025-02-04:
> OpenSCAD <https://www.openscad.org/> version 2024..1.1. (git bc0d078
> )
>
> That looks like media-gfx/openscad-2024.11.14_pre (there's a bug in the
> ebuild breaking the reported Version). Of course there is also
> media-gfx/openscad-2024.12.30_pre available and the live ebuild. Also the
> ebuilds may give some more specific Gentoo info about how to build.
>
> With 0c8ac0e
> I do get a warning dialog box if I do "File/Save As..." to a read-only
> directory and a log message in the console but with a lot of strange
> characters #5637 <https://github.com/openscad/openscad/pull/5637>. Qt
> 6.8.1 and 6.8.2 on GNOME 47.

I've tried other ebuilds in gentoo, all are masked for various reasons.  I
would like to continue trying, too many other issues at the moment.  Tried
to build by compiling, no luck there either.  I have other fires
smoldering, so I will try again in a few days.  Good luck.

Kenny