openscad/openscad #5634: Parameter definition

anoblet, 2025-02-01:
**Is your feature request related to a problem? Please describe.**
```
include <BOSL2/std.scad>

width = 10;
depth = 10;
height = 10;

rounding = 1;

cuboid([width, depth, height], except=[BOTTOM,TOP], rounding = rounding);
```

In this example, `rounding` is better than `rounding = rounding`.

Could we check if a variable is defined in the current scope, and if it is use that value? 

In JavaScript, these to examples are the same:
```

```
function example(arg1, arg2, rounding) {}
```

1. `example(arg1, arg2, rounding = rounding;`

2. `example(arg1, arg2, rounding);`
```

**Describe the solution you'd like**
`example(arg1, arg2, rounding);`

**Describe alternatives you've considered**
I've considered shifting to python, or Rust implementations.

**Additional context**
None :)

jordanbrown0, 2025-02-01:
In JavaScript, `example(arg1, arg2, rounding=rounding)` assigns the value of rounding to itself (effectively doing nothing in most cases), then passes that value to the function for its third parameter.  It is equivalent to
```
rounding = rounding;
example(arg1, arg2, rounding);
```

In OpenSCAD, `example(arg1, arg2, rounding=rounding)` passes `arg1` and `arg2` as the first two parameters, and then passes the current-scope value of `rounding` for the parameter named `rounding`.

You can always pass all parameters positionally, if you want to pass all previous parameters and if you don't feel the need for the mnemonic assistance of specifying the name.  You can also always pass all parameters by name, or a mixture.

For the particular case of BOSL2's `cuboid()`, [their documentation](https://github.com/BelfrySCAD/BOSL2/wiki/shapes3d.scad#module-cuboid) only promises that the `size` parameter can be used positionally; all of the others are available only by name.  You could examine their sources and determine the actual order of the parameters - it happens that `rounding` is the fifth - but they haven't promised that they won't change that order and so your program might break when you upgrade.

t-paul, 2025-02-02:
I'm still not sure what the original question or request is. OpenSCAD has both positional and named parameters, so depending on the function declaration, named parameters allow other parameters to be unspecified and take their default values. That's very much like python handles function parameters too.
```
>>> def fn(a, b = 2, c = 3):
...     print(a, b, c)
...     
>>> fn(1)
1 2 3
>>> fn(1, c=0)
1 2 0
```
For checking if a variable is not defined, there is `is_undef()` - https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Type_Test_Functions#is_undef

I don't see anything actionable here on the OpenSCAD side, if you would like to see changes in BOSL2, then you would need to ask there.

closed 2025-02-02 by t-paul