openscad/openscad #5638: Filenames with umlauts or special chars cannot be rendered on Windows

flesser, 2025-02-01:
**Describe the bug**
If a filename/path contains a German Umlaut or (like `ü`) or esszett (`ß`) it cannot be rendered in OpenSCAD on Windows 10.
The editor is able to read and write those files, but other components like the renderer apparently are not.

Other language-specific special chars are probably affected as well.
This seems to only affect the Windows build of OpenSCAD (and possibly Mac, too? #5551), the Linux build works as expected.

This effectively prevents me from working on any file in my home directory, since my Windows user name contains an `ß`.

**To Reproduce**
Steps to reproduce the behavior:
1. Open new empty file (File -> New)
   - _window title shows "Unbenannt.scad*" (German version of "Unnamed.scad*")_
2. create minimal example geometry, e.g. `sphere(5);` and press F5 / preview
   - _geomery is rendered, but console says "WARNING: Failed to create backup file"_
3. Save file with an umlaut in filename or path, e.g. "Füßbäll.scad"
   - _window title shows "Füßbäll.scad"_
   - _rendered preview vanishes, console says "ERROR: Parser error: file access denied" / "ERROR: Compilation failed! (no top level object found)"_
4. Close OpenSCAD and open it again by double-clicking on "Füßbäll.scad" in Windows Explorer
   - _window title correctly shows "Füßbäll.scad"_
   - _editor correctly shows file content_
   - _preview/render does not work, same errors as in 5._

Additionally there seems to be an issue with how the welcome dialog handles special chars in file names:

5. Close OpenSCAD again, open new empty OpenSCAD window with welcome dialog (e.g. by launching openscad.exe)
6. click on "Füßbäll.scad" in the "last used files" list of the welcome dialog
   - _window title shows garbled encoding of filename_
   - _editor is empty_
   - [image]
   - _rendering works without error (not even the warning from 2.), but of course the scene is as empty as the editor_
 
**Expected behavior**
Files can be opened and rendered regardless of any special chars in the file name or path (as long as they are valid in the OS/filesystem).

**Environment and Version info (please complete the following information):**
 - OS: Windows 10 (German)
 - System: Intel PC 64-bit
 - OpenSCAD-2025.01.31
   - installed from https://files.openscad.org/snapshots/OpenSCAD-2025.01.31-x86-64-Installer.exe
   - defaulting to German localization (inherited from OS settings I guess)

 
**Library & Graphics card information**
seems relevant since even here the `ß` in my home dir is misprinted in two different ways (`E├ƒer` / `E▀er` instead of `Eßer`).

```
C:\Program Files\OpenSCAD (Nightly)>openscad.exe --info

C:\Program Files\OpenSCAD (Nightly)>
QObject::startTimer: Timers can only be used with threads started with QThread
Fontconfig error: Cannot load default config file: No such file: (null)
OpenSCAD Version: 2025.01.31 (git 0c8ac0e6c)
System information: Microsoft Windows 10 (10.0.19045) x86_64 8 CPUs 15.86 GB RAM
User Agent: OpenSCAD/2025.01.31 (git 0c8ac0e6c) (Microsoft Windows 10 (10.0.19045) x86_64)
Compiler: GCC "13.2.0" 64bit
MinGW build: MingW64
Debug build: No
Boost version: 1_81
Eigen version: 3.4.0
CGAL version, kernels: 5.5, Cartesian<Gmpq>, Extended_cartesian<Gmpq>, Epeck
OpenCSG version: OpenCSG 1.5.0
Clipper2 version: 1.5.2
Manifold version: 3.0.1
Qt version: 5.15.11
QScintilla version: 2.11.2
InputDrivers:
GLib version: 2.70.2
lodepng version: 20230410
libzip version: 1.5.2
fontconfig version: 2.14.2
freetype version: 2.13.2
harfbuzz version: 7.3.0
cairo version: 1.16.0
lib3mf version: 1.8.1
Features: roof, input-driver-dbus, lazy-union, vertex-object-renderers-indexing, textmetrics, import-function, predictible-output
Application Path: C:/Program Files/OpenSCAD (Nightly)
Documents Path: C:\Users\Florian E├ƒer\Documents
User Documents Path: C:\Users\Florian E├ƒer\Documents
Resource Path: C:/Program Files/OpenSCAD (Nightly)
User Library Path: C:/Users/Florian E├ƒer/Documents/OpenSCAD/libraries
User Config Path: C:\Users\Florian E├ƒer\AppData\Local/OpenSCAD
Backup Path: C:/Users/Florian E├ƒer/Documents/OpenSCAD/backups
OPENSCADPATH: <not set>
OpenSCAD library path:
  C:/Users/Florian E├ƒer/Documents/OpenSCAD/libraries
  C:/Program Files/OpenSCAD (Nightly)\libraries

OPENSCAD_FONT_PATH: <not set>
OpenSCAD font path:
  C:/WINDOWS/fonts
  C:/Users/Florian E▀er/AppData/Local/Microsoft/Windows/Fonts
  C:/Users/Florian E▀er/.local/share/fonts
  C:/usr/local/share/fonts
  C:/usr/share/fonts
  C:/usr/X11/lib/X11/fonts
  C:/System/Library/Fonts
  C:/Library/Fonts
  C:/Users/Florian E▀er/Library/Fonts

GL context creator: WGL (old)
PNG generator: lodepng

GLEW version: 2.1.0
OpenGL Version: 4.6.0 - Build 27.20.100.9171
GL Renderer: Intel(R) HD Graphics 630
GL Vendor: Intel
RGBA(8888), depth(24), stencil(8)
GL_ARB_framebuffer_object: yes
GL_EXT_framebuffer_object: yes
GL_EXT_packed_depth_stencil: yes
```

pfefferCode, 2025-02-15:
Hi!

I have the same issue.
As a workaround perplexity.ai says - This might even help the developers:

To get OpenSCAD to open paths with umlauts in Windows, you can try the following steps:

Change the code page in your OpenSCAD script file:
Add the following line at the beginning of your OpenSCAD script:

```
system("chcp 1252");
```

This sets the code page to ANSI (1252), which allows for the processing of umlauts.

Save the OpenSCAD file in ANSI format:

Open the .scad file in a text editor like Notepad.

When saving, select "ANSI" as the encoding.

Escape sequences for umlauts:
As an alternative, you can replace umlauts in pathnames with their hexadecimal representation:

    ä = \xE4

    ö = \xF6

    ü = \xFC
    Example: "D:\1. Schule T\xF6ssfeld\Datei1.txt"

BUT: I did not try it because I could easily avoid using umlauts.

Best regards,
   PfefferCode.

t-paul, 2025-02-15:
> BUT: I did not try it because I could easily avoid using umlauts.
Then don't post AI generated suggestion!