openscad/openscad #5639: Could my library be added?

Stone-Age-Sculptor, 2025-02-03:
Hello, could my library be added to https://openscad.org/libraries.html
My library: https://github.com/Stone-Age-Sculptor/StoneAgeLib

t-paul, 2025-02-03:
Done via https://github.com/openscad/openscad.github.com/pull/147.

As this is manually maintained by me, the rules so far have been:
- Generally usable for own projects (e.g. no finished project to "just print")
- Open Source License
- Well documented

While I did not see a user manual, the examples look great, so the library ticks all boxes. Thanks for sharing.

Also posted on [Mastodon](https://fosstodon.org/@OpenSCAD/113941269015236649) and [Bluesky](https://bsky.app/profile/openscad.bsky.social/post/3lhc4zrqxe22w).

closed 2025-02-03 by t-paul

Stone-Age-Sculptor, 2025-02-04:
As you can see, I'm working on it: https://github.com/Stone-Age-Sculptor/StoneAgeLib/wiki