openscad/openscad #5641: Axes shown are not exact in Perspective view

gsohler, 2025-02-03:
**Describe the bug**

When switching to perspective view and pressing the "+X"  button,
I expect the eye to be directly in front of my coordinate system and x axis direcly pointing into my eye.

This is exactly, what is shown in the small coordinate system
Howvwer, I also expect the  big axis to be a point, too, but in fact its shown as a line.
Is this an error or rather inaccuracy in the GL rendering engine ?
(orthogonal view works perfectly

A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. switch to perspecive mode, disable axes labelling
2. Click on X+

**Expected behavior**
X axis shall display as a poiint, only

**Code reproducing the issue**

**Environment and Version info (please complete the following information):**
 any recnt windows build

[image]

kintel, 2025-02-03:
Could you provide Help->Library Info output?
I'm curious which hardware and drivers this is using.

For the record, it works as expected on my machine (macOS), but I haven't tested on Windows native in a while.

gsohler, 2025-02-04:
viewport translatation does not neccessarily have to be 0/0 for perspective view and pressing "X+" and similar.
However very likely some zero's in 

https://github.com/openscad/openscad/blob/master/src/glview/GLView.cc#L397-#L414

neet to get replaced with the actual view translation

kintel, 2025-02-04:
Could you give it a shot fixing it?