openscad/openscad #5642: Incorrect render with Manifold

wiseguy720, 2025-02-04:
**Describe the bug**
I have some code which generates a difference of a cylinder and a union of 5 polyhedrons.  Rendering with CGAL produces the correct output, but rendering with Manifold ignores 4 of the polyhedrons.

**To Reproduce**
Steps to reproduce the behavior:
1) Use the code below
2) Choose CGAL in preferences
3) Clear cache
4) Render (notice the ends of the threads in the nut are beveled)
5) Choose Manifold as the render engine
6) Clear cache
7) Render (notice the ends of the threads in the nut are not beveled - only the difference of the cylinder and first polyhedron are rendered)

**Expected behavior**
Output of CGAL and Manifold should be identical

**Code reproducing the issue**
Sorry this is so large - the code that generated this geometry is over 1600 lines.  I created this testcase from a CSG tree dump simplified it as much as I could.

`union() {
  difference() {
    cylinder($fn = 6, $fa = 12, $fs = 2, h = 6, r1 = 10, r2 = 10, center = false);
    multmatrix([[1, 0, 0, 0], [0, 1, 0, 0], [0, 0, 1, 3], [0, 0, 0, 1]]) {
      union() {
        union() {
          union();
          union();
          multmatrix([[1, 0, 0, 0], [0, 1, 0, 0], [0, 0, 1, -1.75], [0, 0, 0, 1]]) {
            difference() {
              union() {
                polyhedron(points = [[5, 0, -0.65625], [4.1881, 0, -0.1875], [4.1881, 0, 0.1875], [5, 0, 0.65625], [4.33013, 2.5, -0.40625], [3.627, 2.09405, 0.0625], [3.627, 2.09405, 0.4375], [4.33013, 2.5, 0.90625], [2.5, 4.33013, -0.15625], [2.09405, 3.627, 0.3125], [2.09405, 3.627, 0.6875], [2.5, 4.33013, 1.15625], [0, 5, 0.09375], [0, 4.1881, 0.5625], [0, 4.1881, 0.9375], [0, 5, 1.40625], [-2.5, 4.33013, 0.34375], [-2.09405, 3.627, 0.8125], [-2.09405, 3.627, 1.1875], [-2.5, 4.33013, 1.65625], [-4.33013, 2.5, 0.59375], [-3.627, 2.09405, 1.0625], [-3.627, 2.09405, 1.4375], [-4.33013, 2.5, 1.90625], [-5, 0, 0.84375], [-4.1881, 0, 1.3125], [-4.1881, 0, 1.6875], [-5, 0, 2.15625], [-4.33013, -2.5, 1.09375], [-3.627, -2.09405, 1.5625], [-3.627, -2.09405, 1.9375], [-4.33013, -2.5, 2.40625], [-2.5, -4.33013, 1.34375], [-2.09405, -3.627, 1.8125], [-2.09405, -3.627, 2.1875], [-2.5, -4.33013, 2.65625], [0, -5, 1.59375], [0, -4.1881, 2.0625], [0, -4.1881, 2.4375], [0, -5, 2.90625], [2.5, -4.33013, 1.84375], [2.09405, -3.627, 2.3125], [2.09405, -3.627, 2.6875], [2.5, -4.33013, 3.15625], [4.33013, -2.5, 2.09375], [3.627, -2.09405, 2.5625], [3.627, -2.09405, 2.9375], [4.33013, -2.5, 3.40625], [5, 0, 2.34375], [4.1881, 0, 2.8125], [4.1881, 0, 3.1875], [5, 0, 3.65625], [4.33013, 2.5, 2.59375], [3.627, 2.09405, 3.0625], [3.627, 2.09405, 3.4375], [4.33013, 2.5, 3.90625], [2.5, 4.33013, 2.84375], [2.09405, 3.627, 3.3125], [2.09405, 3.627, 3.6875], [2.5, 4.33013, 4.15625], [-5, 0, -0.65625], [-4.1881, 0, -0.1875], [-4.1881, 0, 0.1875], [-5, 0, 0.65625], [-4.33013, -2.5, -0.40625], [-3.627, -2.09405, 0.0625], [-3.627, -2.09405, 0.4375], [-4.33013, -2.5, 0.90625], [-2.5, -4.33013, -0.15625], [-2.09405, -3.627, 0.3125], [-2.09405, -3.627, 0.6875], [-2.5, -4.33013, 1.15625], [0, -5, 0.09375], [0, -4.1881, 0.5625], [0, -4.1881, 0.9375], [0, -5, 1.40625], [2.5, -4.33013, 0.34375], [2.09405, -3.627, 0.8125], [2.09405, -3.627, 1.1875], [2.5, -4.33013, 1.65625], [4.33013, -2.5, 0.59375], [3.627, -2.09405, 1.0625], [3.627, -2.09405, 1.4375], [4.33013, -2.5, 1.90625], [5, 0, 0.84375], [4.1881, 0, 1.3125], [4.1881, 0, 1.6875], [5, 0, 2.15625], [4.33013, 2.5, 1.09375], [3.627, 2.09405, 1.5625], [3.627, 2.09405, 1.9375], [4.33013, 2.5, 2.40625], [2.5, 4.33013, 1.34375], [2.09405, 3.627, 1.8125], [2.09405, 3.627, 2.1875], [2.5, 4.33013, 2.65625], [0, 5, 1.59375], [0, 4.1881, 2.0625], [0, 4.1881, 2.4375], [0, 5, 2.90625], [-2.5, 4.33013, 1.84375], [-2.09405, 3.627, 2.3125], [-2.09405, 3.627, 2.6875], [-2.5, 4.33013, 3.15625], [-4.33013, 2.5, 2.09375], [-3.627, 2.09405, 2.5625], [-3.627, 2.09405, 2.9375], [-4.33013, 2.5, 3.40625], [-5, 0, 2.34375], [-4.1881, 0, 2.8125], [-4.1881, 0, 3.1875], [-5, 0, 3.65625], [-4.33013, -2.5, 2.59375], [-3.627, -2.09405, 3.0625], [-3.627, -2.09405, 3.4375], [-4.33013, -2.5, 3.90625], [-2.5, -4.33013, 2.84375], [-2.09405, -3.627, 3.3125], [-2.09405, -3.627, 3.6875], [-2.5, -4.33013, 4.15625], [5, 0, -1.65625], [4.33013, 2.5, -1.65625], [2.5, 4.33013, -1.65625], [0, 5, -1.65625], [-2.5, 4.33013, -1.65625], [-4.33013, 2.5, -1.65625], [-5, -0, -1.65625], [-4.33013, -2.5, -1.65625], [-2.5, -4.33013, -1.65625], [-0, -5, -1.65625], [2.5, -4.33013, -1.65625], [4.33013, -2.5, -1.65625], [5, 0, 5.15625], [4.33013, 2.5, 5.15625], [2.5, 4.33013, 5.15625], [0, 5, 5.15625], [-2.5, 4.33013, 5.15625], [-4.33013, 2.5, 5.15625], [-5, -0, 5.15625], [-4.33013, -2.5, 5.15625], [-2.5, -4.33013, 5.15625], [-0, -5, 5.15625], [2.5, -4.33013, 5.15625], [4.33013, -2.5, 5.15625], [5, 0, -1.65625], [4.33013, 2.5, -1.65625], [2.5, 4.33013, -1.65625], [0, 5, -1.65625], [-2.5, 4.33013, -1.65625], [-4.33013, 2.5, -1.65625], [-5, -0, -1.65625], [-4.33013, -2.5, -1.65625], [-2.5, -4.33013, -1.65625], [-0, -5, -1.65625], [2.5, -4.33013, -1.65625], [4.33013, -2.5, -1.65625], [5, 0, 5.15625], [4.33013, 2.5, 5.15625], [2.5, 4.33013, 5.15625], [0, 5, 5.15625], [-2.5, 4.33013, 5.15625], [-4.33013, 2.5, 5.15625], [-5, -0, 5.15625], [-4.33013, -2.5, 5.15625], [-2.5, -4.33013, 5.15625], [-0, -5, 5.15625], [2.5, -4.33013, 5.15625], [4.33013, -2.5, 5.15625]], faces = [[4, 0, 1], [6, 1, 2], [7, 2, 3], [1, 5, 4], [1, 6, 5], [2, 7, 6], [8, 4, 5], [10, 5, 6], [11, 6, 7], [5, 9, 8], [5, 10, 9], [6, 11, 10], [12, 8, 9], [14, 9, 10], [15, 10, 11], [9, 13, 12], [9, 14, 13], [10, 15, 14], [16, 12, 13], [18, 13, 14], [19, 14, 15], [13, 17, 16], [13, 18, 17], [14, 19, 18], [20, 16, 17], [22, 17, 18], [23, 18, 19], [17, 21, 20], [17, 22, 21], [18, 23, 22], [24, 20, 21], [26, 21, 22], [27, 22, 23], [21, 25, 24], [21, 26, 25], [22, 27, 26], [28, 24, 25], [30, 25, 26], [31, 26, 27], [25, 29, 28], [25, 30, 29], [26, 31, 30], [32, 28, 29], [34, 29, 30], [35, 30, 31], [29, 33, 32], [29, 34, 33], [30, 35, 34], [36, 32, 33], [38, 33, 34], [39, 34, 35], [33, 37, 36], [33, 38, 37], [34, 39, 38], [40, 36, 37], [42, 37, 38], [43, 38, 39], [37, 41, 40], [37, 42, 41], [38, 43, 42], [44, 40, 41], [46, 41, 42], [47, 42, 43], [41, 45, 44], [41, 46, 45], [42, 47, 46], [48, 44, 45], [50, 45, 46], [51, 46, 47], [45, 49, 48], [45, 50, 49], [46, 51, 50], [52, 48, 49], [54, 49, 50], [55, 50, 51], [49, 53, 52], [49, 54, 53], [50, 55, 54], [56, 52, 53], [58, 53, 54], [59, 54, 55], [53, 57, 56], [53, 58, 57], [54, 59, 58], [3, 2, 1, 0], [56, 57, 58, 59], [64, 60, 61], [66, 61, 62], [67, 62, 63], [61, 65, 64], [61, 66, 65], [62, 67, 66], [68, 64, 65], [70, 65, 66], [71, 66, 67], [65, 69, 68], [65, 70, 69], [66, 71, 70], [72, 68, 69], [74, 69, 70], [75, 70, 71], [69, 73, 72], [69, 74, 73], [70, 75, 74], [76, 72, 73], [78, 73, 74], [79, 74, 75], [73, 77, 76], [73, 78, 77], [74, 79, 78], [80, 76, 77], [82, 77, 78], [83, 78, 79], [77, 81, 80], [77, 82, 81], [78, 83, 82], [84, 80, 81], [86, 81, 82], [87, 82, 83], [81, 85, 84], [81, 86, 85], [82, 87, 86], [88, 84, 85], [90, 85, 86], [91, 86, 87], [85, 89, 88], [85, 90, 89], [86, 91, 90], [92, 88, 89], [94, 89, 90], [95, 90, 91], [89, 93, 92], [89, 94, 93], [90, 95, 94], [96, 92, 93], [98, 93, 94], [99, 94, 95], [93, 97, 96], [93, 98, 97], [94, 99, 98], [100, 96, 97], [102, 97, 98], [103, 98, 99], [97, 101, 100], [97, 102, 101], [98, 103, 102], [104, 100, 101], [106, 101, 102], [107, 102, 103], [101, 105, 104], [101, 106, 105], [102, 107, 106], [108, 104, 105], [110, 105, 106], [111, 106, 107], [105, 109, 108], [105, 110, 109], [106, 111, 110], [112, 108, 109], [114, 109, 110], [115, 110, 111], [109, 113, 112], [109, 114, 113], [110, 115, 114], [116, 112, 113], [118, 113, 114], [119, 114, 115], [113, 117, 116], [113, 118, 117], [114, 119, 118], [63, 62, 61, 60], [116, 117, 118, 119], [120, 0, 4], [4, 121, 120], [121, 4, 8], [8, 122, 121], [122, 8, 12], [12, 123, 122], [123, 12, 16], [16, 124, 123], [124, 16, 20], [20, 125, 124], [126, 60, 64], [64, 127, 126], [127, 64, 68], [68, 128, 127], [128, 68, 72], [72, 129, 128], [129, 72, 76], [76, 130, 129], [130, 76, 80], [80, 131, 130], [39, 141, 142], [142, 43, 39], [43, 142, 143], [143, 47, 43], [47, 143, 132], [132, 51, 47], [51, 132, 133], [133, 55, 51], [55, 133, 134], [134, 59, 55], [99, 135, 136], [136, 103, 99], [103, 136, 137], [137, 107, 103], [107, 137, 138], [138, 111, 107], [111, 138, 139], [139, 115, 111], [115, 139, 140], [140, 119, 115], [3, 84, 88], [88, 7, 3], [7, 88, 92], [92, 11 [... cut ...]
    }
  }
}`

**Screenshots**
CGAL (correct) render: 

[image]

Manifold (incorrect) render:

[image]

**Environment and Version info (please complete the following information):**
 - OS: Windows 11
 - System: Intel PC 64-bit
 - OpenSCAD Version 2025.01.31 (git 0c8ac0e6c)

**Library & Graphics card information**
If applicable, add the content of the Help->Library Info window, or the output of running the `openscad --info` command.

**Additional context**
Add any other context about the problem here.

pca006132, 2025-02-04:
Your 4 polyhedrons are flipped. The vertices should be in CCW direction when viewed from the outside.

closed 2025-02-04 by pca006132

wiseguy720, 2025-02-04:
Thank you for your reply.  Per the documentation, polyhedrons must be specified in a CW direction when viewed from the outside, but your comment led me to investigate my example further and I see that my code incorrectly specified the faces in the CCW direction.