openscad/openscad #5647: Application Error - Memory could not be read. - When closing a new or recent file

Midknight, 2025-02-05:
Recently updated to the latest development snapshot(s) and the last couple have had the same issue. I would either open a recent file through the menu or just open a new file. I would work on the file, save and then click the close button on the tab of the open file and I would get the following Application Error. The instruction at 0x00007FF7D7A36329 referenced memory at 0x0000000000000080. The memory could not be read. Click on OK to terminate the program.

**To Reproduce**
Steps to reproduce the behavior:
1. Open a recent file or new file.
2. Click the close button of the table of the file you opened
3. See error (Application Error. The instruction at 0x00007FF7D7A36329 referenced memory at 0x0000000000000080. The memory could not be read. Click on OK to terminate the program.)

 - OS: Windows 10
 - System: AMD Ryzen 7 4800HS with Radeon Graphics/Nvidia GeForce RTX 2060
 - OpenSCAD Version 2025.02.05 & 2025.02.01 downloaded from the openscad.org website

OpenSCAD Version: 2025.02.05 (git 34c0e2af7)
System information: Microsoft Windows 10 (10.0.19045) x86_64 16 CPUs 15.42 GB RAM
User Agent: OpenSCAD/2025.02.05 (git 34c0e2af7) (Microsoft Windows 10 (10.0.19045) x86_64)
Compiler: GCC "13.2.0" 64bit
MinGW build: MingW64
Debug build: No
Boost version: 1_81
Eigen version: 3.4.0
CGAL version, kernels: 5.5, Cartesian<Gmpq>, Extended_cartesian<Gmpq>, Epeck
OpenCSG version: OpenCSG 1.5.0
Clipper2 version: 1.5.2
Manifold version: 3.0.1
Qt version: 5.15.11
QScintilla version: 2.11.2
InputDrivers: 
GLib version: 2.70.2
lodepng version: 20230410
libzip version: 1.5.2
fontconfig version: 2.14.2
freetype version: 2.13.2
harfbuzz version: 7.3.0
cairo version: 1.16.0
lib3mf version: 1.8.1
Features: roof, input-driver-dbus, lazy-union, vertex-object-renderers-indexing, textmetrics, import-function, predictible-output
Application Path: C:/Program Files/OpenSCAD (Nightly)
Documents Path: C:\Users\user\Documents
User Documents Path: C:\Users\user\Documents
Resource Path: C:/Program Files/OpenSCAD (Nightly)
User Library Path: C:/Users/user/Documents/OpenSCAD/libraries
User Config Path: C:\Users\user\AppData\Local/OpenSCAD
Backup Path: C:/Users/user/Documents/OpenSCAD/backups
OPENSCADPATH: <not set>
OpenSCAD library path:
C:/Users/user/Documents/OpenSCAD/libraries
C:/Program Files/OpenSCAD (Nightly)\libraries

OPENSCAD_FONT_PATH: <not set>
OpenSCAD font path:
C:/Windows/fonts
C:/Users/user/AppData/Local/Microsoft/Windows/Fonts
C:/Users/user/.local/share/fonts
C:/usr/local/share/fonts
C:/usr/share/fonts
C:/usr/X11/lib/X11/fonts
C:/System/Library/Fonts
C:/Library/Fonts
C:/Users/user/Library/Fonts
GLEW version: 2.1.0
OpenGL Version: 4.6.13596 Compatibility Profile Context 20.10.20.10 27.20.11020.10001
GL Renderer: AMD Radeon(TM) Graphics
GL Vendor: ATI Technologies Inc.
RGBA(8888), depth(0), stencil(8)
GL_ARB_framebuffer_object: yes
GL_EXT_framebuffer_object: yes
GL_EXT_packed_depth_stencil: yes

Qt graphics widget: QOpenGLWidget
QSurfaceFormat: RGBA(8888), depth(0), stencil(8)
[... 282 lines cut ...]
GL_KHR_no_error
GL_KHR_parallel_shader_compile
GL_KHR_robust_buffer_access_behavior
GL_KHR_robustness
GL_KTX_buffer_region
GL_NV_alpha_to_coverage_dither_control
GL_NV_blend_square
GL_NV_conditional_render
GL_NV_copy_depth_to_color
GL_NV_copy_image
GL_NV_depth_buffer_float
GL_NV_explicit_multisample
GL_NV_float_buffer
GL_NV_half_float
GL_NV_primitive_restart
GL_NV_shader_atomic_int64
GL_NV_texgen_reflection
GL_NV_texture_barrier
GL_OES_EGL_image
GL_SGIS_generate_mipmap
GL_SGIS_texture_edge_clamp
GL_SGIS_texture_lod
GL_SUN_multi_draw_arrays
GL_WIN_swap_hint
WGL_EXT_swap_control

kintel, 2025-02-05:
Does this happen only when closing the application or also when closing one of multiple open tabs?

Midknight, 2025-02-06:
This only happens with the very first tab as other tabs it doesn't trigger when I close them. If I where to close the application it doesn't trigger. If I used the File Menu -> Quit route it doesn't trigger.

Midknight, 2025-02-07:
Uninstalled 2025-02-05 and installed the latest nightly 2025-02-07 and still have the issue. 

Are there cache/temp files I should remove before installing? Can you list the file path if so?