openscad/openscad #5649: [Help] The command "openscad mymodule.scad-o mymodule.ast" does not produce the expected AST

huskier, 2025-02-07:
**Environment**
[OpenSCAD](https://www.openscad.org/) version 2021.01

**Issue**
Hi there,

I'm encountering an issue with the openscad command. When I run "openscad mymodule.scad -o mymodule.ast", I do not get the expected Abstract Syntax Tree (AST) as the result.

Here is the code for which we want to get the AST:

```
module mymodule() {
    translate([10, 0, 0]) cube([5, 5, 5]);
    rotate([45, 0, 0]) sphere(10);
}

difference() {
    mymodule();
    cylinder(h = 20, r = 5);
}
```

After running the command "openscad mymodule.scad -o mymodule.ast", the output is the same as the input text:
```
module mymodule() {
    translate([10, 0, 0]) cube([5, 5, 5]);
    rotate([45, 0, 0]) sphere(10);
}
difference() {
    mymodule();
    cylinder(h = 20, r = 5);
}
```
However, we were expecting the result to be something like this:
```
difference
├── module_call: mymodule
│   ├── translate
│   │   ├── [10, 0, 0]
│   │   └── cube
│   │       ├── [5, 5, 5]
│   │       └── {}
│   └── rotate
│       ├── [45, 0, 0]
│       └── sphere
│           ├── 10
│           └── {}
└── cylinder
    ├── h = 20
    └── r
```

Could you please help us understand why the command does not produce the expected AST and how we might resolve this issue?

jordanbrown0, 2025-02-07:
It's not a graphical representation of the AST.  It's a dump of the AST... in OpenSCAD syntax.

If you look carefully you'll find that it's a reconstruction, not the original.  It happens that your original is in exactly the form that the dump reproduces.  If, for instance, you made one of the constants be 5.000, you would see that the AST dump reports it as 5.

This output is the same output that you get in the graphical interface from Design/View AST.

closed 2025-02-07 by jordanbrown0

huskier, 2025-02-08:
@jordanbrown0 Thank you for your explanation.

Indeed, we aim to manipulate the AST to translate the syntax into another language, such as TypeScript. I believe the best approach might be to work with the source code of OpenSCAD. Could you please provide some suggestions? Where or which files/classes should we focus on?

jordanbrown0, 2025-02-08:
The generated AST exists only as an in-memory data structure.  The only thing you can extract is that OpenSCAD-syntax dump.

The lexical analysis is in
https://github.com/openscad/openscad/blob/master/src/core/lexer.l
and the parsing is in
https://github.com/openscad/openscad/blob/master/src/core/parser.y

That might get you to the point of understanding the language syntax, but won't help with getting the semantics right.  The semantics are scattered all across src/core.

huskier, 2025-02-08:
@jordanbrown0  Thank you very much for your quick reply.

We could first handle things inside OpenSCAD and later consider separating them.

Could we directly manipulate the in-memory AST data structure and perform the translation? By doing so, would we be able to accurately capture the semantics?

I wonder where it would be appropriate to dive into the code.

kintel, 2025-02-08:
I see three main avenues of exploration to be able to process OpenSCAD source code separately from OpenSCAD:
1. Export to csg. This gets rid of most complex OpenSCAD constructs and yields a concrete CSG tree. Still in OpenSCAD format, but a small subset of it + other tools like FreeCAD already ingests it.
2. Turn OpenSCAD's AST into a separate component/API. This was initiated in #1743, as a separate C++ library. This still needs a bit of work.
- 3. If you're not into C++, use a separate OpenSCAD parser. We started one at https://bitbucket.org/openscad/openscad-js - this needs a lot more work though.

It all depends on what you want to achieve. Please share more info, preferably on the mailing list or irc, to reach more people.

huskier, 2025-02-08:
@kintel Thank you for your suggestion. I've posted a message titled "About Automatically Translating OpenSCAD Scripting Code into CadQuery Scripting Code (Python Code)" to the mailing list.