openscad/openscad #5651: Special character á in path causes failure to load .scad files

ilykos, 2025-02-08:
**Describe the bug**
On Windows, error 

> "Parsing design (AST generation)...
> ERROR: Parser error: file access denied
> ERROR: Compilation failed! (no top level object found)" 

prevents file from loading and previewing when there is a special character in the path to this file  path. Errors in nightly version, but works in the website release.

**To Reproduce**
Steps to reproduce the behavior:
1. Add special character "á" to you path (preferably parent folder) and try opening a file in 

**Expected behavior**
A clear and concise description of what you expected to happen.

**Code reproducing the issue**
If applicable, add trimmed down, self-contained code demonstrating the issue.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment and Version info (please complete the following information):**
 - OS: Windows 10
 - System: Intel PC 64-bit
 - OpenSCAD Version 2025.02.08, 2024.10.31, or 2025.01.03

**Library & Graphics card information**

OpenSCAD Version: 2024.10.31 (git 5dd397d7e)
System information: Microsoft Windows 10 (10.0.19045) x86_64 8 CPUs 127.91 GB RAM
User Agent: OpenSCAD/2024.10.31 (git 5dd397d7e) (Microsoft Windows 10 (10.0.19045) x86_64)
Compiler: GCC "13.2.0" 64bit
MinGW build: MingW64
Debug build: No
Boost version: 1_81
Eigen version: 3.4.0
CGAL version, kernels: 5.5, Cartesian<Gmpq>, Extended_cartesian<Gmpq>, Epeck
OpenCSG version: OpenCSG 1.5.0
Qt version: 5.15.11
QScintilla version: 2.11.2
InputDrivers: 
GLib version: 2.70.2
lodepng version: 20230410
libzip version: 1.5.2
fontconfig version: 2.14.2
freetype version: 2.13.2
harfbuzz version: 7.3.0
cairo version: 1.16.0
lib3mf version: 1.8.1
Features: roof, input-driver-dbus, lazy-union, vertex-object-renderers-indexing, textmetrics, import-function, predictible-output

[...]

C:/usr/local/share/fonts
C:/usr/share/fonts
C:/usr/X11/lib/X11/fonts
C:/System/Library/Fonts
C:/Library/Fonts
C:/Users/imisystem/Library/Fonts
GLEW version: 2.1.0
OpenGL Version: 4.6.0 NVIDIA 560.94
GL Renderer: NVIDIA GeForce RTX 2080 Ti/PCIe/SSE2
GL Vendor: NVIDIA Corporation
RGBA(8888), depth(24), stencil(8)
GL_ARB_framebuffer_object: yes
GL_EXT_framebuffer_object: yes
GL_EXT_packed_depth_stencil: yes

Qt graphics widget: QOpenGLWidget
[... 380 lines cut ...]
GL_NV_transform_feedback
GL_NV_transform_feedback2
GL_NV_uniform_buffer_std430_layout
GL_NV_uniform_buffer_unified_memory
GL_NV_vertex_array_range
GL_NV_vertex_array_range2
GL_NV_vertex_attrib_integer_64bit
GL_NV_vertex_buffer_unified_memory
GL_NV_vertex_program
GL_NV_vertex_program1_1
GL_NV_vertex_program2
GL_NV_vertex_program2_option
GL_NV_vertex_program3
GL_NV_viewport_array2
GL_NV_viewport_swizzle
GL_OVR_multiview
GL_OVR_multiview2
GL_S3_s3tc
GL_SGIS_generate_mipmap
GL_SGIS_texture_lod
GL_SGIX_depth_texture
GL_SGIX_shadow
GL_SUN_slice_accum
GL_WIN_swap_hint
WGL_EXT_swap_control

pfefferCode, 2025-02-15:
I guess, this is the same bug as https://github.com/openscad/openscad/issues/5638

t-paul, 2025-02-15:
Closing as duplicate of #5638.

closed 2025-02-15 by t-paul