openscad/openscad #5657: Export PDF issues with page orientation and paper size

JonMcLin, 2025-02-09:
**Describe the bug**
Page orientation only Portrait.  Selection of "auto" and explicit selection of "Landscape" does not work.
Paper size only A4 (default).  Selection of "Letter" does not work.  Likely this resulted from removing Qt dependencies since those used Qt enums.

**To Reproduce**
Steps to reproduce the behavior:
1. Generate a 2D shape:  square([250,200]);     // 10"x8"; this should fit on a letter-size sheet in landscape mode, but not in portrait
2. Click on "Render"
3. Select "Export PDF"
4. Select paper "Letter"
5. Select orientation "Auto", or orientation "Landscape".
6. Click okay.
7. Note the "EXPORT WARNING" in the console:  "Geometry is too large to fit into selected size."
8. View the resulting PDF.   Page will be portrait, and image will not be fully displayed.
 
**Expected behavior**
No Export Warning (since I wrote and test the original code for paper size and orientation control and use them often, I know this is the expected behavior.  :)
Image is Landscape oriented.
Image fits in the display.
Image has "Letter" page dimensions.

**Code reproducing the issue**
See above.  That one square([250,200]); produces an appropriate image.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment and Version info (please complete the following information):**
 - OS: Ubuntu 24.04
 - System:  Beelink SER5 AMD Ryzen7
 - OpenSCAD Version 2025.01.26, Ubuntu Store Nightly installer.

**Library & Graphics card information**
If applicable, add the content of the Help->Library Info window, or the output of running the `openscad --info` command.

**Additional context**
I will work on this although my time may be limited for the next couple of months.  I'll need to get my Dev environment set up again.

JonMcLin, 2025-02-10:
Confirmed same issue on Windows 10 

Environment and Version info (please complete the following information):

    OS: Windows 10.0.19045
    System: Lenovo AMD Ryzen7
    OpenSCAD Version 2025.01.15 (git 8f49712f7).

Noticed that this also does not populate the file extension ".pdf".   

Also, walking through the new Settings code (Settings.cc) I found a likely error in line 313 (typo that would not necessarily preclude compilation), but might result in other unexpected behavior of the export PDF.   Once I restore a build environment I'll investigate more.

t-paul, 2025-02-10:
Yep, I'll have a look at the extension issue separately. Closing this via #5662.

closed 2025-02-10 by t-paul

JonMcLin, 2025-02-11:
Awesome!   Thanks for addressing it so quickly.  In the interest of scratching my current itch I just tested it.  (Turned out my environment was only a year or so behind...)

commit 39de7704 (2025-02-10): Merge pull request #5662 from openscad/fix-pdf-export-dialog

Fix PDF export dialog (fixes #5657)