openscad/openscad #5661: import of two items

bamboe32, 2025-02-10:
When importing 2 different .stl files only the first item is show twice.

Windows 11
OpenSCAD version 2025.02.10
Intel PC-64

import("D:/item1.stl");
translate ([0,50,0]) import("D:/item2.stl");

After moving item2.stl to the first line, it still show item2 twice.
After closing/opening OpenSCAD it then sows item2.stl twice

kintel, 2025-02-10:
Can you try putting the STL files in the same folder as your .scad file, and remove the `D:/` prefix?

bamboe32, 2025-02-10:
ok, then it works.
But with the 2021.01 version it works with the directories.
Putting it to my .scad files directory would give a lot of pollutions.

So it still should be treated as a bug.

kintel, 2025-02-10:
Is your `D:` drive on some network share or cloud drive or smth.?

bamboe32, 2025-02-11:
No, it is a normal SSD Drive.
In my case the files are not directly in the root, but is some /sub/sub/sub directory

spegelius, 2025-08-16:
I have this same problem, versions 2025.08.15 and 2025.05.02 tested in Windows 10. Happens when the scad and stl files are located in network drive but not when the files are on hdd. In both cases, the files are located on the same drive, just different directories.
In linux, no issues (2025.08.15) using same network location.

jordanbrown0, 2025-08-17:
I betcha this is a duplicate of #5425 .

BeardyBarber, 2025-09-03:
I just encountered this issue on OpenSCAD version 2025.07.25 (git 6a8ab04bf) and 2025.09.02 (git a16fcd05e)
For me it is blocking ability to work with more then one imported STL file.

**Preconfitions:**
Folder with 2 STL files exists in drive D:
e.g 
D:\3d\STL\file_1.stl
D:\3d\STL\file_2.stl

**Reproduction Steps:**
1. Open new file
2. Import STL file_1 using 
`import("D:\\3d\\STL\\file_1.stl")`
3. Refresh view (F5) or Render (F6)
4. Change imported file to file_2
`import("D:\\3d\\STL\\file_2.stl")`
5. Refresh view (F5) or Render (F6)
6. Save file in folder D:\3d
7. Change import to use relative path
`import("STL\\file_2.stl")`
8. Refresh view (F5) or Render (F6)

**Results:**
"file_1" is only file that user is able to import (creating new file does not fix the issue), 
Issue is fixed only after closing OpenSCAD but still able to import only one file.