openscad/openscad #5667: `for` loops with ranges using very large numbers silently runs unpredictably

stockholmux, 2025-02-12:
**Describe the bug**
When using very large numbers exceeding the precision of the Number type (aka Double) in a range with a for loop, the number of executions becomes silently unpredictable.

**To Reproduce**
Run the following code:

```
rangeTest = [36342608503257320:36342608503257320];

for(i= rangeTest) {
    echo("run");
}
```
produces (predictably):
```
ECHO: "run"
```

However, when you increment the `end` in the range by one, starts to get odd:
```
rangeTest = [36342608503257320:36342608503257321];

for(i= rangeTest) {
    echo("run");
}
```
It also produces:
```
ECHO: "run"
```
(should be `ECHO: "run"` twice)

Now, changing the end to `36342608503257324`:

 ```
rangeTest = [36342608503257320:36342608503257324];

for(i= rangeTest) {
    echo("run");
}
```
I would expect this would `ECHO: "run"` 5x, but it outputs 9x:
```
ECHO: "run"
ECHO: "run"
ECHO: "run"
ECHO: "run"
ECHO: "run"
ECHO: "run"
ECHO: "run"
ECHO: "run"
ECHO: "run"
```

(I understand why this is happening)

**Expected behavior**

When Numbers exceed the presentable precision it's probably okay in many contexts that exist in OpenSCAD, however in Ranges this produces a very difficult to debug issue.

OpenSCAD should probably throw at least a warning if any part of a range exceeds the precision of a Number/Double. (Although it might be nice to throw warnings anytime Numbers exceed the Double precision, but that is a bigger issue)

**Code reproducing the issue**
See above.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment and Version info (please complete the following information):**
 - OS: MacOS
 - System: ARM
 - OpenSCAD Version [e.g. 2021.01, 2023.04.16] and where it was installed from: Nightly version 2025.02.10 (git 48c04246a)

**Library & Graphics card information**
n/a

**Additional context**
n/a

jordanbrown0, 2025-02-12:
As it happens, my bit-wise arithmetic wad includes a warning if an integer constant cannot be represented precisely.

That doesn't help if a calculated value exceeds the available precision. Also note that you don't need big numbers to exceed available precision - 0.1 cannot be represented precisely in binary  floating point no matter how many bits you use.

jordanbrown0, 2025-02-12:
Note also that this range behavior is almost the same behavior that you would get if you tried to step from 1.0 to 1.00000000000000010 by  0.0000000000000001.  (Approximately. I'm not at a computer and can't double check.)

stockholmux, 2025-02-12:
@jordanbrown0 I'm actually watching that Bitwise PR very closely. 

My interest and concern is primarily about big integer numbers: they silently start failing in unexpected ways (e.g. in my `for` example).

IMHO, with very tiny float representation you can bound the behaviour in more predictable ways (and most developers have experienced all sort of floating point shenanigans).

jordanbrown0, 2025-02-12:
The behavior is actually exactly the same for both very large and very small numbers.  You get a precision of about one part in 10^16.

There are few reasons for people to notice the issues with very large numbers, because they generally just aren't very practical.  10^16 is approximately enough to model the entire Solaris System at one-millimeter resolution.

If you're just trying to pack a whole lot of bits of data into a single number, I submit that you're using the wrong data representation.

closed 2026-10-21 by github-actions[bot]