openscad/openscad #5670: import of SVG with viewbox attribute overrides centering

wolfwood, 2025-02-14:
via a reddit post https://www.reddit.com/r/openscad/comments/1in51qq/comment/mcaw2im/ by @garblesnarky

`import("test.svg", center=true);`

with [test.svg](https://github.com/user-attachments/assets/83782e33-94f8-4fba-9515-9d17521f398a)

doesn't result in the imported file living at the origin. the user determined a viewbox attribute is the reason.

I was assuming this was a bug, but from the svg centering code it could be intentional? I'm looking for clarification so I can add an appropriate testcase, and a fix (or not).

wolfwood, 2025-02-15:
this is where the viewbox is used to translate each point.
https://github.com/openscad/openscad/blob/master/src/io/import_svg.cc#L203-L204

wolfwood, 2025-02-18:
it looks like originally SVG centering worked as expected. this commit https://github.com/openscad/openscad/commit/d5b3ae2c158fc3b1dd5f9ba16ca61993b4f7b8fb added scaling based on the viewbox. it doesn't explicitly mention centering behavior.

there's no test of the interaction between viewbox and center. removing `-viewbox.x()` and `-viewbox.y()` from the centering here https://github.com/openscad/openscad/blob/master/src/io/import_svg.cc#L203-L204 still passes all tests.

t-paul, 2025-02-19:
Right, the "auto-center" of the test suite hides placement issues when there is not another fixed object added.

I suppose the common expectation would be to still scale based on the viewbox, but center on actual geometry? The current calculation for the center case seems to be wrong in any case, so we should fix that.

wolfwood, 2025-02-19:
that matches my expectation. I updated my PR for this, #5683, to fix the calculation.

stefano2734, 2026-09-02:
Not so important for next OpenScad Release, when more than a year untested with actual dev versions?

Alexb715, 2026-11-12:
i'll start working on this