openscad/openscad #5672: Make interactive measurement tool release ready

kintel, 2025-02-16:
An interactive measurement tool was added in #4813.
The tool works reasonable well, but it has some weaknesses causing some user frictions:

* It only works after rendering (F6) in Manifold mode. .onsider extend this feature to support CGAL mode as well, as we're still going to ship CGAL rendering.
* In the cases where it doesn't work (preview), it's a bit hard to discover why.
* It's not obvious how to disable the annotation of a previously made measurement

kintel, 2025-05-07:
@coryrc Yes, this is the overall ticket to discuss if anything needs to be done to the measurement tool to make it into the next release. We may split topics from this ticket into blockers vs. optional Feature Requests, but a discussion comes first.

gsohler, 2025-09-09:
> Still planning to work on this.

One very suitable improvement would be to turn the buttons into toggle
buttons.
This makes it very intuitive to cancel measurement prematurely

https://imgur.com/a/59fs6Zm

coryrc, 2025-09-09:
@gsohler What do you think the user journey would be?

i.e. move mouse around, highlight appears, click to lock in that highlight, then... what? same as now? continuously-updating measurement in a window? only pop up when you click another thing?

then... are you saying it un-toggles once you click again? or if you click the toggle to end prematurely?

coryrc, 2025-11-14:
Additionally, I have found some existing issues:

* #5931 (possibly solvable by [#5960](https://github.com/openscad/openscad/pull/5960))
* #5732 (already fixed -- though it suggests a different format for output, which may or may not be useful)

coryrc, 2025-11-27:
#6299 fixed a bunch grab-bag of edge case bugs causing incorrect measurements.

Günther suggested copying additional measurements from PythonScad, including face-to-edge. I'm not sure that should block the release. I'll have to check if it is easy.

Still intend to have the measured values written to the console. At that point I think the tool is release-ready and we can close.

kintel, 2025-11-30:
Adding more features just makes it harder, so I'd say write up a ticket but not necessarily make it blocking.

coryrc, 2026-01-04:
Does anyone know why this LOG statement prints into the console:

https://github.com/openscad/openscad/blob/6cad20bd4ae15bff2e307afc4e330aaa81659c61/src/RenderStatistic.cc#L328

But this LOG statement, which is immediate line prior to calling the function the previous LOG is in, does not?

https://github.com/openscad/openscad/blob/6cad20bd4ae15bff2e307afc4e330aaa81659c61/src/gui/MainWindow.cc#L1552

kintel, 2026-01-04:
There may be some threading challenges in determining in which MainWindow to print the log statement.
I think some of this could be refactored, as this is very old code:
https://github.com/openscad/openscad/blob/6cad20bd4ae15bff2e307afc4e330aaa81659c61/src/gui/MainWindow.cc#L3656-L3666

t-paul, 2026-01-04:
Yep, the output handling probably belongs into something similar to the EvaluationSession but with lifetime of the whole processing.

coryrc, 2026-01-05:
> Yep, the output handling probably belongs into something similar to the EvaluationSession but with lifetime of the whole processing.

Or maybe I can call `void Console::addMessage(const Message& msg)` via `this->console.addMessage()` in the measurement click handler?

coryrc, 2026-01-05:
Oh, hello:

```
void MainWindow::consoleOutput(const Message& msgObj, void *userdata)
{
  // Invoke the method in the main thread in case the output
  // originates in a worker thread.
  auto thisp = static_cast<MainWindow *>(userdata);
  QMetaObject::invokeMethod(thisp, "consoleOutput", Q_ARG(Message, msgObj));
}
```

coryrc, 2026-01-05:
Though, 1, I don't see that being used, 2, I think it's in the main thread because it's a GUI connected method.

coryrc, 2026-01-05:
Okay, I took four measurements in a row:

```
Rendering finished.
Distance is 30.5666
dx: -5.85786  dy: 0  dz: 30
Angle is 70.391 Degrees
Distance is 15.1185
Distance is 19.1421
Point to Line Endpoint1: dx: -12.9289  dy: 14.1421  dz: 0
Point to Line Endpoint2: dx: -20  dy: 7.07107  dz: 0
Perpendicular distance to (infinite) line: dx: 13.5355  dy: -13.5355  dz: 0
```

Does anybody want to see dividers between these? I could stick some `===` on a line before each measurement?

coryrc, 2026-01-06:
Here's a question: What is `[9, 11, -5.414]` ?

jordanbrown0, 2026-01-06:
> No, it's a sample input for one of those programming puzzles that asks how many times you cross zero while summing numbers.

I don't think that's ever appeared in my OpenSCAD console log.  I must not be running the right models.

> The only time is when you're going back sometime later, when you'll want added context, and it's mixed with everything else that ends up in the console.

And at that point all you have is ... a measurement.  No context saying what was being measured.

I'm still not sure that adding the "Distance" label adds much value.  But at least it's almost minimal noise, so I don't object to it.

---

I didn't say anything about vertexes; I just said a mark.  Where the mark is, is your business.

But:  there are places where you can't readily place a mark, because you don't know their location until after you do the boolean operations.  So it's not a panacea.

coryrc, 2026-01-07:
Well, measurements between line segments are not correct, at least in corner cases. Using the above case, the distance is calculated as `19.97106876373290874`, but the distance between the two closes endpoints is `14.736 230 39` so there's something more seriously wrong here.

Also measuring between these points:

```
  dx: -12.9289  dy: 7.07107  dz: 0
  First selection point is at [-20, -7.0710678100585938, 30]
  Second selection point is at [-32.928932189941406, 0, 30]
```

Gives NaN for the distance, despite being quite readily calculable. Hmm.

coryrc, 2026-01-08:
I couldn't figure out how to fix it, so I generated something based on Dan Sunday's algorithm and it passed the tests.