openscad/openscad #5675: Color inheritance glitchyness on sibling geometries

FourCinnamon0, 2025-02-17:
**Describe the bug**
the `color()` function seems to oddly apply to shapes before bitwise operations (the colour from an unrelated geometry is applied to another geometry). 

(see screenshot)

**To Reproduce**
1. Render code in PREVIEW mode
2. Observe

**Expected behavior**
The solid at the top should be fully blue without the weird grey glitchyness, and only the bottom thin cylinder should be grey.

**Code reproducing the issue**
```scad
layer_h=4;

translate([0,-0.1,0]) color([0.4,0.4,1])
difference(){
    translate([40,0,0])
    cylinder(100,r=50);
    translate([0,0,layer_h/2-5]) cube([1000,1000,layer_h+10],center=true);
};

translate([0,-0.1,-10]) color([0.4,0.4,0.4])
intersection(){
    translate([40,0,0])
    cylinder(100,r=50);
    translate([0,0,layer_h/2]) cube([1000,1000,layer_h],center=true);
};
```

**Screenshots**
[image]

**Environment and Version info:**
 - OS: Microsoft Windows 11 Pro
 - System: x64-based PC, Intel64 Family 6 Model 141
 - OpenSCAD Version 2024.12.04 (git c190dacd1) (installed from website) (originally)
 - also reproduced on OpenSCAD/2025.02.16 (git 615787a1a)

**Library & Graphics card information**
```
MinGW build: MingW64
Debug build: No
Boost version: 1_81
Eigen version: 3.4.0
CGAL version, kernels: 5.5, Cartesian<Gmpq>, Extended_cartesian<Gmpq>, Epeck
OpenCSG version: OpenCSG 1.5.0
Clipper2 version: 1.5.2
Manifold version: 3.0.1
Qt version: 5.15.11
QScintilla version: 2.11.2
InputDrivers: 
GLib version: 2.70.2
lodepng version: 20230410
libzip version: 1.5.2
fontconfig version: 2.14.2
freetype version: 2.13.2
harfbuzz version: 7.3.0
cairo version: 1.16.0
lib3mf version: 1.8.1
Features: roof, input-driver-dbus, lazy-union, vertex-object-renderers-indexing, textmetrics, import-function, predictible-output
Application Path: C:/Program Files/OpenSCAD (Nightly)
```

**Additional context**
N/A

jordanbrown0, 2025-02-17:
Coincident surfaces - surfaces that are precisely aligned with one another - are known to produce preview artifacts, even when they are from negative objects or pieces of objects that are differenced or intersected away.

This is probably covered under issue #1000.  Also #4826 and probably others.

Try to avoid having coincident surfaces.  Wrapping a subassembly in `render()` may help, but can introduce preview performance issues and convexity issues.  It used to lose color, but it looks like that's been fixed when using Manifold.

kintel, 2025-02-17:
This isn't really about coincident surfaces; it's a core weakness of the preview algorithm where independent intersections and differences influence each other. As Jordan points out, a good workaround is to place render() calls around the difference() and intersection() calls.

This is a clean duplicate for #1000, so I'll close this and keep tracking this issue there.

closed 2025-02-17 by kintel