openscad/openscad #5676: OpenScad v 2025.01.26 (Snap on Linux Mint) - Save File As: Can't navigate to mounted NAS locations (/mnt/...) (on NAS)

dbyy, 2025-02-17:
**Describe the bug**
When opening the dialog "Save As" to save a script on a mounted drive (/mnt/nas_data/projects/) I cannot navigate to the desired location. 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'File'
2. Click on 'Save As'
3. Navigate to '/mnt/'
4. no directories are visible

**Expected behavior**
I would expect to see all the directories located in /mnt/

Drop-Down-List "File Types" allows only the selection of type *.SCAD. Check mark "Show hidden files" is selected.

[image]

 - OS: Linux Mint 21.3
 - System: AMD Ryzen 3-3200G]
 - OpenSCAD Version [2025.01.16 - snap] 
 

This functionality is available in OpenSCAD v2024.08.18.ai20248 (git a16ca2a67)

t-paul, 2025-02-18:
As far as I know, this is not possible automatically with Snap as we run the sandboxed setup and that is not going to change.

It might be possible to use the "removable-media" interface (see https://forum.snapcraft.io/t/interface-management/6154) but I'm not sure how that works and people seem to have mixed results. It seem that is only targeting USB drives and such devices.

dbyy, 2025-02-18:
> As far as I know, this is not possible automatically with Snap as we run the sandboxed setup and that is not going to change.
>
> It might be possible to use the "removable-media" interface (see [...]

That means I can save my scripts only in my $HOME location, nowhere else. 
I would think that's limiting the use of OpenSCAD on Linux drastically, even only in the SNAP environment.
I just downloaded the Appimage (same version). In this case I can store my scripts on mounted NAS locations.
The experts may explain why this is possible in the AppImage but not in the Snap environment. I'm not one of them.

t-paul, 2025-02-18:
Here is the wrong place for that discussion. This is a property of Snap packaging defined by Ubuntu, nothing we can change. We provide Snap packages for people who like the benefits and can live with the restrictions imposed by Ubuntu (mostly for security reasons I assume). If those restrictions don't work for you, I suggest using a different packaging. We provide many options for exactly that reason.
I really would prefer one simple packaging working everywhere, that would save me many hours of fighting with those different formats. But unfortunately no such solution is in sight at the moment.

dbyy, 2025-02-19:
no blame on you and OpenSCAD :-)

closed 2025-02-19 by dbyy