openscad/openscad #5682: Background and Debug modifier transparency differ in non-obvious way.

pipatron, 2025-02-19:
Now that https://github.com/openscad/openscad/issues/3689 is closed I can report this minor transparency preview bug with a simple example. I don't think this works as intended but I'm not 100% sure.

In short: The background and debug modifier's transparency are different in a non-obvious way.

A simple example without any intersections, differences, overlapping objects:
````
translate([0,0]*20)  cube(15);
translate([1,0]*20) %cube(15);
translate([2,0]*20) #cube(15);
translate([1,1]*20)  cube(15);
translate([2,1]*20) %cube(15);
translate([0,1]*20) #cube(15);
translate([2,2]*20)  cube(15);
translate([0,2]*20) %cube(15);
translate([1,2]*20) #cube(15);
````
[image]

I expect both debug objects and background objects to be transparent, and they are. I also expected to see debug objects behind background objects, and vice versa. However, there are differences:

- Good: Normal objects shows through both debug and background objects.
- Good: Background objects (grey) are visible through the debug objects (red).
- Bad: Debug objects (red) are blocked/invisible behind background objects (grey).

I don't see the logic behind this, so I assume it's a bug, albeit minor. I'm leaving it here as a trivial example since it may help someone else debug one of the other remaining transparency issues, if any. The example is specifically designed to remove order-dependency as in this bug: https://github.com/openscad/openscad/issues/1390 (I think).

> **[Background modifier](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Modifier_Characters#Background_modifier)**
>
> Ignore this subtree for the normal rendering process and draw it in transparent gray (all transformations are still applied to the nodes in this tree).

> **[Debug modifier](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Modifier_Characters#Debug_modifier)**
>
> Use this subtree as usual in the rendering process but also draw it unmodified in transparent pink.

OpenSCAD 2025.02.18.nightly (git 91070a0) on Debian 12 (bookworm) with GNOME 43.9 on Wayland

kintel, 2025-02-19:
This is probably a duplicate of #1390

jordanbrown0, 2026-05-08:
I think it's #1390 combined with the fact that highlight and background are split out into their own rendering chains and so the program doesn't control the ordering between them.