openscad/openscad #5684: Problem exporting STL file in OpenSCAD 2019.05

0v3rcl0ck3d, 2025-02-19:
Hi,
I am pretty new to this. I have an issue with the generated STL file.
Steps I did:
- Download scad file here: https://www.thingiverse.com/thing:2404395/files
- open the file in OpenSCAD 2019.05
- Change the parameters
- preview
- render (F6)
- export STL file
- import STL file to Chitubox

Expected result:
Preview in Chitubox looks like the preview in OpenSCAD

Observed result: see screenshots

[image]

[image]

Here is the scad file:
[Customizable_Card_Box.scad.zip](https://github.com/user-attachments/files/18873422/Customizable_Card_Box.scad.zip)

This is the generated STL file. I am sure that I did render before generating the STL file. And I already used many different STL files in Chitubox to print them.
[Card_Box_exploding_kittens.stl.zip](https://github.com/user-attachments/files/18873347/Card_Box_exploding_kittens.stl.zip)

If you need further information, please let me know. I would appreciate your help. Thanks in advance.

jordanbrown0, 2025-02-19:
2019.05 is really old. Please upgrade to at least 2021.01 (most recent official release) or a development snapshot. See the Downloads link at OpenSCAD.org.

t-paul, 2025-02-19:
Apart from the old version, the STL looks fine in Prusa-Slicer (also note the "No errors detected", so it did not try to auto-repair the STL). I suppose this issue should be posted against Chitubox.

[image]

0v3rcl0ck3d, 2025-02-19:
I installed the newer version of OpenSCAD 2021.01 and it works. However, I am a bit surprised that Prusa does not report any errors. Problem solved, thanks.

t-paul, 2025-02-19:
That is indeed strange. I guess that proves what Greg Kroah-Hartman postulated quite some time ago: "[Computers Barely Work](https://www.youtube.com/watch?v=t9MjGziRw-c)" 🤣.