openscad/openscad #5685: A backspace as first character gets wrong in the string.

Stone-Age-Sculptor, 2025-02-19:
Suppose this string:
```
string = "\bABC";
```
In the 'C' language, a backspace as the first character is ignored when the string is printed. The string length is 4, but only "ABC" is printed. In the first position is the backspace, which is character 8.
In OpenSCAD, the string becomes "bABC". The string length is also 4, but the first character is not 8, but 98 for the 'b'.

Tested with OpenSCAD 2025.02.19

```
string = "\bABC";
echo(string=string,len=len(string),codes=ord(string[0]),ord(string[1]),ord(string[2]),ord(string[3]));
```

Output: `ECHO: string = "bABC", len = 4, codes = 98, 65, 66, 67`

t-paul, 2025-02-19:
> But this test can check for "\b":

No, it does not check for "\b" as backspace, it checks for "b" = ascii 98.

Stone-Age-Sculptor, 2025-02-19:
Thank you for the explanation. I understand now that the 'if' statement just found the "b".
I was implementing the backspace for my own text function, but I have to remove that.

t-paul, 2025-02-19:
To support extended features we'll need the top level of the text rendering stack. Right now, we only have Harfbuzz for text shaping, but lacking the layouting (e.g. Pango). I hope we'll get that eventually, but it's not a small topic.

For a great overview on that topic, see the articles by Behdad Esfahbod (author of Harfbuzz) 
- https://behdad.org/text/
- https://behdad.org/text2024/

jordanbrown0, 2025-02-20:
OpenSCAD strings do not support as many backslash codes as C strings do.

See [the manual](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/General#Strings) for a list of supported backslash codes.

OpenSCAD does *not* support:
* `\a` - alert (bell)
* `\b` - backspace
* `\f` - formfeed
* `\v` - vertical tab

Once you get the character into a string, *then* you can start talking about whether text() and echo() do anything interesting with that character.  And the answer there is mostly "no".

Stone-Age-Sculptor, 2025-02-20:
I'm working on my own text function: https://github.com/Stone-Age-Sculptor/StoneAgeLib/wiki/Subdivision-Font
The UTF-8 standard allows private characters, for example in the range U+E000 ... U+F8FF.
Instead of using one of the backslash escapes, I could make a private UTF-8 character that says, for example: write upside down from now on.