openscad/openscad #5687: Preference "Always show 3mf exporter dialog" doesn't update when you disable the dialog from its own window

ali1234, 2025-02-20:
**Describe the bug**
When you export as 3mf a dialog is shown. It has a checkbox at the bottom. If you uncheck it, the dialog is no longer shown. There is another checkbox in preferences which does the same thing, however these two check boxes are not synced.

**To Reproduce**
Steps to reproduce the behavior:
1. Make a cube or something.
2. File -> export -> 3mf
3. Uncheck "Always show this dialog" at the bottom.
4. Complete the export.
5. Edit -> Preferences -> Dialogs
6. "Always show 3mf export dialog" will be ticked.
7. File -> Export -> 3mf (the dialog is not shown)
8. Edit -> Preferences -> Dialogs
9. Untick the checkbox
10. Tick the checkbox
11. File -> Export -> 3mf
12. Now the dialog is shown

**Expected behavior**
The checkbox in preferences should reflect the current state of the setting, regardless of where the change was made.

**Code reproducing the issue**
N/A

**Screenshots**

[image]

[image]

**Environment and Version info (please complete the following information):**
 - Xubuntu 24.10, Openscad Nightly (snap) 2025.01.20.ai22731 (git 0272f796a)

kintel, 2025-02-20:
Reproducible also on macOS.

t-paul, 2025-02-20:
Oh, I guess I always tested with closing the application between. In that case it works. But as there are multiple places in the app, the preferences dialog needs to re-init those settings when the dialog opens.

NOTE: You can also get the dialog once when pressing SHIFT key when starting the export. I'm not yet sure how to convey that info to the user. Maybe as text in the preferences dialog?

ali1234, 2025-02-20:
The exact same thing happens with the PDF dialog of course.

And another paper cut:

1. File -> Export -> 3mf
2. Untick "Always show dialog"
3. Click cancel
4. File -> Export -> 3mf

Result: the dialog is shown.

But:

1. File -> Export -> 3mf
2. Untick "Always show dialog"
3. Click "Okay"
4. Click "cancel" on the file requester
5. File -> Export -> 3mf

Result: the dialog is not shown.

This can be sidestepped by showing the filename requester first, and the export settings second. This is the approach GIMP uses when exporting, for example.