openscad/openscad #5689: projection() takes a very long time on some imported STLs

ali1234, 2025-02-21:
**Describe the bug**
projection() takes a very long time (possibly forever) on certain STL files generated by OpenSCAD.

**To Reproduce**
Steps to reproduce the behavior:
1. Open shape.scad and export as STL (ascii).
2. Open proj.scad.

**Expected behavior**
The preview should not take forever.

**Code reproducing the issue**

File: shape.scad
```
$fa = 1;
$fs = 0.1;

rotate([0, -90, 90])
rotate([-2, 0, 0])
linear_extrude(height=2.5)
offset(2) offset(4) {
    offset(6) offset(-6) square([54, 39], center=true);
    square([23, 53], center=true);
}
```

File: proj.scad
```
projection() import("shape.stl");
```

**Environment and Version info (please complete the following information):**
Xubuntu 24.10, Nightly 2.19 (appimage)

**Additional context**
It happens with both manifold and cgal.
If you remove the outermost rotation from shape.scad and instead apply it in proj.scad, the problem still happens.
If you make pretty much any other change to shape.scad, the preview completes in a fraction of a second with manifold, about 1 second with cgal.

jordanbrown0, 2025-02-21:
The import seems unnecessary; wrapping projection() around the original object (around the outer rotate()) yields the pathological case.

I see the problem only with Manifold, not with CGAL.  (Don't forget to flush caches when switching.)

This is with 2024.12.26 on Windows.

Perhaps the stacked offsets are generating a huge number of tiny segments.

jordanbrown0, 2025-02-21:
It did finally complete, after ~7m.

When I unioned the base object (without the projection()) with a cube(), CGAL said:
>WARNING: Object may not be a valid 2-manifold and may need repair!

It had ~6000 facets and ~4000 vertexes.

Preview with Show Edges shows a lot of very small triangles:

[image]

The lighter areas are densely packed very narrow triangles.

Zooming in on the top right corner...

[image]

jordanbrown0, 2025-02-21:
And those thicker diagonal lines are themselves very narrow triangles:

[image]

pca006132, 2025-02-21:
Ah, this is triggering the slow case for triangulation in manifold.

pca006132, 2025-02-21:
For some reason it treats the polygon as something with many holes, and the keyhole function is taking a long time.

pca006132, 2025-02-21:
I think there are two problems here:

1. Manifold projection may produce bad geometry. For openscad, the simplest way to fix this might be to use the other backend to do the projection until this is fixed upstream.
2. Manifold triangulation is slow when you have thousands of holes in the polygon... This is probably not that important but still something the upstream should probably look into later.

elalish, 2025-02-22:
Hmm, Manifold's [`Project()`](https://manifoldcad.org/docs/html/classmanifold_1_1_manifold.html#a328657b69e9bbaaee148572ca314677c) method doesn't do any triangulation. It only returns the polygons, which it says should be run through a positive rill rule pass as they are likely to contain lots of self-intersections. Is it possible that step is being skipped in OpenSCAD? 

Or more likely (since the result is lots of holes), is it using the even-odd fill rule instead? I've noticed from some of the test cases in OpenSCAD that it seems to like defaulting to even-odd instead of positive, for reasons I cannot comprehend. Even-odd fill rule will give pathological results in pretty much all cases.

pca006132, 2025-02-22:
OpenSCAD does pass the polygons to `CrossSection` with the positive fill rule (default), but it doesn't perform simplification on it. I tried simplification, it helps in some cases, but for the complex case here it doesn't work.

elalish, 2025-02-22:
Yeah, given the crazy amount of degenerate geometry in this object, I'd say this is fairly expected behavior, but we should look and see if the simplification pass can be improved. Perhaps this shape has accrued more rounding error than it's reporting. 

Hmm, is OpenSCAD using manifold's exact rotation transformations? This shape looks like it should have a lot of exactly vertical triangles, which should give face normals with exactly 0 zero Z components. Any floating point error in those Z values will cause this ugliness, but Manifold's transforms are designed to be exact for 90-degree rotations for exactly this reason. Is OpenSCAD introducing any rounding error, or is Manifold's face normal calculation unstable?

pca006132, 2025-02-23:
The mesh is rotated by 2 degrees, so we cannot use the exact transform.

ali1234, 2025-02-23:
Yes, that 2 degree rotation was a necessary part of the original thing I was making. It is basically flat with a sloped top. I removed all the unnecessary parts and just the sloped part was left. I can show the original if you want - it takes an unknown amount of time to process, and I also saw occasional segfaults but I couldn't reproduce those easily.

elalish, 2025-02-24:
Well, the 2 degrees is around the Z-axis, then 90's around the Y and then Z again. That should leave the original top and bottom surfaces of the extrusion exactly vertical, right? I'm curious if that's precisely true.

pca006132, 2025-02-24:
I looked at the code, openscad is maintaining its own transform matrix, so it is not using manifold's rotation function.

nophead, 2025-02-24:
I think OpenSCAD has had exact transforms for 90 degrees for a long time.

elalish, 2025-05-11:
I just converted this example into a pure ManifoldCAD.org version: https://tinyurl.com/79weamet

It's working no problem in 0.03 seconds. If anyone can manipulate my example into something that shows the slowness, I'll be happy to investigate. However, I believe the problem is rounding error introduced in steps outside of the Manifold library.

pca006132, 2025-08-06:
or maybe this is an artifact from ear-clipping? The outer triangles in the CGAL triangulation are mostly degenerates...

elalish, 2025-08-06:
was there intended to be an image included above? I'm not quite following the logic here. How is this triangulation related? I thought the problem was all with vertical faces getting projected down to a line, and floating-point issues creating lots of cruft that's slow to sort through.

pca006132, 2025-08-06:
CGAL's triangulation:

[image]

Our triangulation:

[image]

It is not related to our projection code. If we export the mesh produced by manifold (with the triangulation in the second image), import it, and do a project with manifold disabled, there is the same issue.
Instead, if we export the first one, import it, and do a project with manifold disabled, it works fine. (probably because most triangles are simplified to nothing by clipper).

kintel, 2025-08-13:
The initial example seems to work well with Manifold as of v3.2.1 - can we close this as fixed, or are there other corner cases which should be highlighted in this issue?

elalish, 2025-08-14:
Glad to hear it - if anyone finds a corner case with a Manifold-only repro, please file it upstream and we'll take a look.