openscad/openscad #5696: Colored console text output

amatulic, 2025-02-24:
**Problem statement**

OpenSCAD is capable of outputting console text with different background colors. Errors are reddish, warnings are yellow, trace output is light blue.

* As a developer, I would like to be able to highlight certain console output messages in a different background color to be more noticeable to users.

* As a user, I would like to see non-error non-warning notifications from libraries I use in a different color so I can distinguish them from my own debugging console output.

The only related issue I could find is #2607 from 2018 - but that's all about HTML support, which no longer exists for console output. That issue should probably be closed.

**Solutions**

Only one alternative listed below would need to be implemented, whichever is simplest.

1. Make the existing color() statement be capable of modifying console output background color. For example:
`color("lightgreen") text("This is a notification with a light green background.");`

2. Give the echo() statement color arguments. Example:
`echo("This is a notification", bgcolor="lightgreen", fgcolor="black");`
I'd be happy with just `bgcolor`.

3. Introduce a new statement notify() that works just like echo() but outputs a hardcoded background color different from errors (red), warnings (yellow), and traces (blue). Maybe cyan or light green.
`notify("This text uses the internal 'notification' color.");`

**Alternatives considered**
* ANSI escape sequences for color -- not supported
* HTML markup -- no longer supported
* seeing if `color()` can modify echo output -- didn't work but didn't generate an error either

jordanbrown0, 2025-02-24:
Tidbit:  `color(...) echo(...);` doesn't generate an error because it isn't an error.  The `color()` module changed the color of all of the geometry generated by its children... and it doesn't mind that its children didn't generate any geometry.

I don't think that `color()` can be the right answer, because changing the color of text and changing the color of geometry would interact strangely.

I'd like to see HTML-subset processing return, though probably through a separate mechanism; an `echo()` caller shouldn't need to be aware of HTML quoting requirements.

ANSI sequences are just too obscure.

Arguments to `echo()` like you describe seem tempting, but can't work because `echo()` already has semantics for named arguments.  Your example above yields:

> ECHO: "This is a notification", bgcolor = "lightgreen", fgcolor = "black"

amatulic, 2025-02-24:
Whoops, you're right, I forgot about the feature of echo supporting named arguments. Okay, scratch that.

I wouldn't mind having some sort of tag inside the string to set a color, doesn't have to be HTML, but I agree the user shouldn't need to be aware of HTML quoting.

The more I think about it, I really would be satisfied with a separate "notification" color. This could be done with another form of echo like `notify()` that outputs a color like `assert()` already outputs red.

Or for that matter, assert() could have a "level" argument, causing it to output different colors, and not aborting processing for level 0... I'll modify the issue description above.