openscad/openscad #5700: Import JSON, Spaces in key names

thestumbler, 2025-02-25:
**Describe the bug**
If the JSON file has field keys with spaces in them, how do you access that field with the syntax provided in the users manual?

**To Reproduce**
```
flags = import("assy.json");
if(flags==undef) {
  echo("NO JSON FILE");
} else {
  echo(flags);
  echo(flags.fileFormatVersion);
  echo(flags.parameterSets);
}
```

**Expected behavior**

I don't know what to expect. The method of accessing JSON structures as described in the manual uses dotted notation. The example "people.json" file, data is accessed by t.people, person.name, etc. That notation doesn't permit spaces in the key name.

I did try flags.parameterSets.New\ set\ 1, flags.parameterSets[0], and len(flags.parametersSets) (thinking I could iterate over them), and neither of those ideas worked.

**Code reproducing the issue**
see above

**Environment and Version info (please complete the following information):**
 - OS: MacOS Sequoia 15.3.1, 
 - System: MBP 2021 M1
 - OpenSCAD Version: 2025.01.27 (git 11c07a0d6) from Developer snapshots on website

**Additional context**
Why am I doing this? I was bothered by the following behavior: when I changed variables for the customizer, I kept being asked if I wanted to save when I closed the GUI (I edit my file in an external editor). I assumed that this was wanting to update the .SCAD file itself, with new values as set in the customizer (this may be a wrong assumption). My idea was to not put any customizer variables in the SCAD file at all, and they would only exist in the JSON file. Not surprising, this simplistic approach did not work. So I tried to add an initialization block at the top of my file, so that if JSON existed, it would load and set the customizer variables. If it didn't exist, I would set those variables to default values.

This was when I figured out the default parameter set names had spaces in them, like "New set 1", "New set 2", etc.

jordanbrown0, 2025-02-25:
It's Javascript-like.  For constant identifier-compatible names you can use dot notation; you can index by a string for non-constant and non-identifier cases.

`flags["parameterSets.New set 1"]`

jordanbrown0, 2025-02-25:
Sorry, didn't read the question quite right.  Should be
`flags.parameterSets["New set 1"]`

(I'm editing that into the original answer and deleting this followup.)

jordanbrown0, 2025-02-25:
Your assumption about being asked to save on quit is close.  It doesn't rewrite the `.scad` file, but it does rewrite the `.json` file.

closed 2025-02-25 by jordanbrown0

thestumbler, 2025-02-25:
So, upon open, it will load the JSON file automatically? 
If I want the hard-coded defaults to take effect again, I just delete the JSON file?

I wonder if an additional phrase might be included in the pop-up message that explains to the user which file the program wants to save.

jordanbrown0, 2025-02-25:
I'm not an expert on the behavior of the Customizer, but what I observe is that it reloads the JSON file and defaults to the "<design default>" parameter set, the values specified in the `.scad` file.

jordanbrown0, 2025-02-25:
Argh.  Not quite plain text.  Should have said

defaults to the `<design default>` parameter set