openscad/openscad #5709: 3D Navigation style

erjoalgo, 2025-02-26:
I am used to the blender navigation style and find it difficult to switch back and forth between openscad and the blender 3D navigation. For example, in blender orbiting and panning is done via the middle click and middle click + shift. But in the openscad 3D viewer, this would be left click, right click.

It would be useful to support adjusting the navigation style. Other users have expressed the same desire:

[https://www.reddit.com/r/openscad/comments/kzpurw/blender_style_view_navigation_with_the_mouse/](url)

> Is there a way in OpenSCAD to change the view navigation style to the style that Blender uses (similar to how FreeCAD lets you set this)? Did a brief search of this, and I'm not sure if this is supported. I use Blender as well as FreeCAD, and it's been really helpful in FreeCAD to be able to use the same keys and mouse buttons to move around. I'm using version 2019.05 (Ubuntu).

**Is your feature request related to a problem? Please describe.**
N/A

**Describe the solution you'd like**
A "navigation style" dropdown menu option, where users can select among a few different navigation styles. Each navigation style would likely only require re-binding the orbiting and panning mouse bindings.

Freecad implements this feature, here's how it looks like as an example:

[image]

A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
Alternatively, exposing the key bindings used for orbiting and panning as a menu option would also suffice.

**Additional context**

harrybraviner, 2025-03-05:
You can at least switch the orbiting and panning mouse bindings with this button:

[image]

It sounds like matching Blender would require enabling the middle button as a pan and/or rotate control?

I would find this feature useful, as I regularly switch between OpenSCAD and Fusion, and it would help my brain to be able to use a common navigation style for each. If we added a page to Preferences that enabled selection of the actions performed by `[modifier] + [mouse button]`, with a drop-down menu to set them to defaults (OpenSCAD, Blender, Fusion), would other people also find that useful? I've been poking around at the code and would be happy to implement this.

t-paul, 2025-03-05:
I suppose the ideal solution would be unifying the mouse input with the input control code for 3d mouse / gamepad / joystick. But that also might be one of those "too big to do in one step" cases.

So starting with a configurable mouse behavior might be a good first step towards more options for users who are accustomed to other mouse setups.

harrybraviner, 2025-03-05:
I've been poking around at this and I think I might be able to put something together that works, but isn't too elegant. If I can do so and send you a branch from my fork, would you be willing to provide feedback?

t-paul, 2025-03-05:
Sure, if you prefer it this way. I can pull from a branch to have a look and give it a spin.

harrybraviner, 2025-03-07:
@t-paul If you want to pull from this branch and give it a try, I think it should work: https://github.com/harrybraviner/openscad/tree/mouse_styles

I added a preferences page where you can select from preset mouse behaviour:

[image]

Or you can just select 'Custom' and pick whatever behaviour you want. Note that I haven't tidied up my changes, and this is my first time really adding a UI feature to anything, so try not to judge my code quality too harshly :-)

@erjoalgo I believe I've correctly replicated the behaviour of Blender. If you want to pull from my branch and try it out, you can let me know if that's what you're after.

erjoalgo, 2025-03-26:
@harrybraviner I tested your branch today and it works well, it does feel the same as the navigation in Blender. Thanks for the great work and quick response.

[image]

t-paul, 2025-03-26:
Oh, thanks for the ping and the test. I totally missed the earlier comment even though I usually follow the mail notifications. I'm going to do some traveling in the upcoming days but I'll try to make some time.

t-paul, 2025-03-27:
On initial run, it seems to default to all actions = 0, so only the scroll wheel works. I guess there's 2 options.
1) have the detail actions ignored if a scheme != custom is set
2) have everything default to OpenSCAD actions (so nothing in config file means unchanged behavior)

In general I think that looks nice. For looking through the code it would help a lot if there's a PR as that easily shows the changes and gives a place to comment on specific code places.

harrybraviner, 2025-04-24:
@t-paul - my apologies for sitting on this for a few weeks.

Happy to make a PR, though I want to tidy up a little first. I also want to fix this issue you describe: "On initial run, it seems to default to all actions = 0". I'm going to try and fix this over the next few days, but I'm a bit concerned that each time I recompile and run, I may be picking up some cached settings from the previous time I compiled & ran. Does OpenSCAD persist settings within the binary, or to somewhere else? (In the latter can I would have to clear those to reproduce your issue.)

t-paul, 2025-04-24:
No worries with the timeline. OpenSCAD moves at volunteer speed, meaning things happen when people can make the time.

Yes, settings are persisted via Qt (QSettings) so on Linux that is usually `$HOME/.config/OpenSCAD/OpenSCAD.conf`. Windows uses the Registry. Removing this file should catch pretty much all of the automatic persistent information. For reproducing the issue you have to delete those settings, I believe.

Additionally there's a runtime cache for the model which can be flushed via Design -> Flush Caches but that should not impact anything related to the topic here.

harrybraviner, 2025-05-01:
Ok, I've fixed this so that the behaviour on first start-up is the OpenSCAD default navigation style. I've also cleaned up the code and opened a PR: #5874

t-paul, 2025-05-01:
Nice, thank you! I'll have a look and try to poke people for checking it out on platforms other than Linux.