openscad/openscad #5718: Projection segfault

ali1234, 2025-03-01:
**Describe the bug**
OpenSCAD segfaults when taking the projection of a particularly complex object. Possibly related to #5689 - it is the same object that originally exposed that issue. The segfault happens with a git master build. It does not happen with AppImages.

**To Reproduce**
Steps to reproduce the behavior:
1. Get https://al.zerostem.io/~al/openscad/bbox-test/hingeplate.stl
2. Run the SCAD code.

**Expected behavior**
No segfault.

**Code reproducing the issue**
```
projection(cut = false) 
multmatrix([ [ 0, 1, 0, 0 ], [ 0, 0, 1, 0 ], [ 1, 0, 0, 0 ], [ 0, 0, 0, 1 ] ]) 
import("hingeplate.stl");
```

**Environment and Version info (please complete the following information):**
 - OS: Xubuntu 24.10
 - System: AMD 9950x
 - OpenSCAD Version: git master, self built.

**Library & Graphics card information**
Attached for crashing and not-crashing versions.

[crash.txt](https://github.com/user-attachments/files/19035154/crash.txt)

[nocrash.txt](https://github.com/user-attachments/files/19035155/nocrash.txt)

Here's the diff between them:

```

< OpenSCAD Version: 2025.02.28.ai23859 (git ce6161cc0)
---
> OpenSCAD Version: 2025.03.01
3,4c3,4
< User Agent: OpenSCAD/2025.02.28.ai23859 (git ce6161cc0) (Linux x86_64; Ubuntu 24.10)
< Compiler: GCC "11.4.0" 64bit
---
> User Agent: OpenSCAD/2025.03.01 (Linux x86_64; Ubuntu 24.10)
> Compiler: GCC "14.2.0" 64bit
6,7c6,7
< Debug build: No
< Boost version: 1_74
---
> Debug build: Yes
> Boost version: 1_83
9,11c9,10
< CGAL version, kernels: 5.5.1, Cartesian<Gmpq>, Extended_cartesian<Gmpq>, Epeck
< Python Version: Python3.10.12
< OpenCSG version: OpenCSG 1.7.0
---
> CGAL version, kernels: 5.6.1, Cartesian<Gmpq>, Extended_cartesian<Gmpq>, Epeck
> OpenCSG version: OpenCSG 1.5.0
14,15c13,14
< Qt version: 5.15.3
< QScintilla version: 2.11.6
---
> Qt version: 5.15.15
> QScintilla version: 2.14.1
17c16
< GLib version: 2.72.4
---
> GLib version: 2.82.1
20,23c19,22
< fontconfig version: 2.13.1 (runtime: 2.15.0)
< freetype version: 2.11.1 (runtime: 2.13.3)
< harfbuzz version: 2.7.4 (runtime: 9.0.0)
< cairo version: 1.16.0
---
> fontconfig version: 2.15.0
> freetype version: 2.13.3
> harfbuzz version: 9.0.0
> cairo version: 1.18.2
25,26c24
< Features: roof*, input-driver-dbus, lazy-union, vertex-object-renderers-indexing, textmetrics, import-function, predictible-output, python-engine
< Application Path: /tmp/.mount_OpenSCbOBeaK/usr/bin
---
> Application Path: /home/al/Source/OpenSCAD/openscad/build
29c27
< Resource Path: /tmp/.mount_OpenSCbOBeaK/usr/share/openscad
---
> Resource Path: /home/al/Source/OpenSCAD/openscad
36c34
< /tmp/.mount_OpenSCbOBeaK/usr/share/openscad/libraries
---
> /home/al/Source/OpenSCAD/openscad/libraries
89c87
< GLAD version: 2.0.4
---
> GLEW version: 2.2.0
503c501
< GL_SUN_slice_accum
\ No newline at end of file
---
> GL_SUN_slice_accum
```

ali1234, 2025-03-01:
It seems that the segfault only happens with debug builds. The same exact source built as release does not crash, and just takes forever to process the projection.

rcolyer, 2025-03-01:
I confirm.  I was unable to reproduce this segfault with any build or AppImage I tried until this debug insight, and then tried a debug build and this produces the segfault.

kintel, 2025-03-01:
For reference: It crashes using the OpenSCAD-2025.02.28 macOS official snapshot release build.

kintel, 2025-11-01:
Minimal reproducible test case:
```
polygon(points=[
[-0.9009814262390137,4.396420694887638],
[-0.9009196311235428,4.396418534219265],
[-0.9009199738502502,4.396418549120426],
[-0.9009201154112816,4.396418549120426],
[-0.900920145213604,4.396418556571007],
[-0.9007904008030891,4.396414026618004],
[-0.900920532643795,4.396418564021587],
[-0.9009205549955368,4.396418571472168],
[-0.8948094919323921,4.39620516449213],
[-0.9009838402271271,4.396420776844025],
[-0.9007960110902786,4.396414220333099],
[-0.900975875556469,4.396420501172543],
[-0.9009200632572174,4.396418549120426],
[-0.900920107960701,4.396418549120426],
[-0.900923028588295,4.3964186534285545],
[-0.9009454548358917,4.396419435739517],
[-0.9007934629917145,4.396414130926132],
[-0.9007934853434563,4.396414130926132],
[-0.9008689969778061,4.3964167684316635],
[0.08876987546682358,1.8603339567780495]
]);
```

kintel, 2026-01-04:
Fixed in Clipper2 V2.0.1.
Will bump, and verify in a snapshot build before closing this.

closed 2026-01-14 by kintel