openscad/openscad #5719: Allow for range notation in arrays and strings to extract sub-arrays and substrings

amatulic, 2025-03-01:
**Problem description**
As a user of OpenSCAD, I have found a need to extract substrings from strings.

**Solution I'd like**
Allow array and string indexing to use range notation. For example:
* Given `a = [ 1, 2, [3,4], 5, [6,7,8]];` I would like to use `a[1:3]` to extract the sub-array `[2, [3,4], 5]`.
* Given `s = "Now is the time";` I would like to use `s[4:8]` to get the substring `"is the"`.

A high bound beyond the length of the array or string would be clamped at the maximum index of the array or string, the low bound would be clamped at 0, and if the entire range is out of range, the return would be an empty array or empty string.

**Alternatives considered**
One can do something similar with submatrix() and substr() function calls provided in BOSL2, but being able to use OpenSCAD's existing range notation would be more elegant, and just makes sense. Since I started using OpenSCAD 5 years ago, this has seemed like an obvious missing feature. Also, I imagine the function calls have significantly more overhead than if OpenSCAD extracted the array/string range internally.

jordanbrown0, 2025-03-01:
Note that for arrays this is relatively easy using list comprehension:
```
a = [ 1, 2, [3,4], 5, [6,7,8]];
b = [ for (i = [1:3]) a[i] ];
```

And you could DIY a slice function that would be fairly fast.  (Not as fast as built-in, but at least not recursive concatenation.)
```
function slice(a, r) = [ for (i = r) a[i] ];
b = slice(a, [1:3]);
```

But alas there is no comprehension mechanism for strings.  A join() function would be helpful there, ref #4450.

jordanbrown0, 2025-03-01:
Note that array-range-notation would not behave the way you might expect for multidimensional arrays.

You might think that given a 2D array `a`, `a[3:4][5:6]` would extract a 2x2 array, but instead would extract an array consisting of two arrays, and would then attempt (and fail) to extract the sixth and seventh elements from that array.

t-paul, 2025-03-01:
See also #3115. That PR wanted too much (include half open ranges / step != 1) so it got stuck. It should be fine salvaging the simple slicing case.

amatulic, 2025-03-02:
Thanks @t-paul , I searched for an issue before writing this one but I didn't find that. I agree, there are way too many features in that PR, and I find the notation like [:<-2] confusing and more than most people would need. Just using the existing range notation would be sufficient.

@jordanbrown0 -
> You might think that given a 2D array a, a[3:4][5:6] would extract a 2x2 array

Actually I'd think it would extract the equivalent of `(a[3:4])[5:6]`, which would be an empty array. But that's a good point, I didn't think of extracting a matrix from a matrix. The way I described it above, you'd have to make an array using `[a[3][5:6], a[4][5:6]]`, which admittedly looks clunky.

rcolyer, 2025-03-02:
I endorse this proposal.  I think this has strong merit as a feature, and that it is palpably missing.  Also, here is a reference implementation in user space that I think does the right things:
```
function slice(vs, r) = is_string(vs) ?
  let(join = function(v, s="", i=0) i>=len(v) ? s : join(v, str(s, v[i]), i+1))
  join([for (i=r) vs[i]]) :
  [for (i=r) vs[i]];

echo(slice("foobar", [1:2:6]));
echo(slice([1,4,9,16,25,36], [1:2:6]));
s = "Object=Grass";
index = search("=", s)[0] + 1;
echo(slice(s, [index:len(s)-1]));
```

Half-open ranges would also be nice, but I suggest that is separate.  This should simply do what ranges already do.  Updating ranges to be half-open could be separate, and likewise with other wishlists like negative indexing for indices and ranges.  If someone wants to tackle it all at once that would be fine, but it's often much simpler and more likely to be successful to work on separable chunks.