openscad/openscad #5721: DEPRECATED: Using ranges of the form [begin:end] with begin value greater than the end value is deprecated...

andyDoucette, 2025-03-02:
I'm getting this a lot these days, and it's super annoying.  I'm either looking for a easy fix, or asking humbly for it to be un-depriciated if we can't find one.

Here's my use case:

I'd like to iterate over a vector that may have any number of elements (including 0).

I want to write:

`for (a=[0:len(my_vector)-1])`, and that used to work, and now it doesn't when my_vector=[].

My current work around is to do this:

```
for (a=[0:len(my_vector)])
    if a!=len(my_vector)
```
But that's super ugly to put everywhere.

This also applies to list comprehensions.

I think iterating over a 0-length vector should produce a 0-length iteration (no-op), and doing a list comprehension over a 0 length vector should result in `[]`.  Makes sense to me.  Thoughts?

t-paul, 2025-03-02:
That never worked, hence the deprecation warning.

Use explicit step of 1 to prevent the warning and handle empty vectors as expected: `[0:1:len(v)-1]`.

As the deprecation exists for quite some time, it might be an option to change the legacy behavior, @kintel?

andyDoucette, 2025-03-02:
Oh, yeah making it work would be great!  Thanks for the easy work-around.  Adding a ":1", while a little less succinct than it could be, is a lot better than my existing work-around.  Thanks for your response! :)

kintel, 2025-03-02:
> As the deprecation exists for quite some time, it might be an option to change the legacy behavior, [@kintel](https://github.com/kintel)?

I haven't thought about this in a long time. Do you think silently turning the deprecated case into an empty range makes sense? ..or issue a warning that the range was negative and thus empty?

andyDoucette, 2025-03-03:
I think it's all about what the programmer intended.  If we go from complaining that their intent was wrong to actually doing what they said, that's only an improvement.  I'm pretty sure that programmers who write something that evaluates to [0:-1] expect it to result in [] (once they know that openSCAD uses inclusive indexing, unlike python which has an exclusive end point).  I'd vote for just doing the "expected thing" and getting rid of the depreciation warning entirely.  

Now the only thing to be aware of is cases where the programmer didn't actually KNOW what they were doing but it JUST WORKED, and it JUST WORKED out of shear luck.  In this case, when they wrote [0:-1] and it was interpreted as [-1:0], and they got 2 elements instead of 0, and they happend to want 2 elements and didn't look close enough at their code to notice that the code wasn't written right.  We'd be breaking their code.  But honestly, if it "just works" but the never understood why it was working, I'd say it never really actually worked at all - we'd be doing them a favor by breaking their code as they'd have to actually understand what the hell they're doing for once to make it work!  hahah :)  Anyway, I very rarely get code to actually work unless I fully understand what I'm doing, and I think that's the case for most people.  So this will affect a tiny percentage of users whereas a hundred times more people are affected when the code doesn't do what they expect it to.  So in all, I think it's the right call.

Being very explicit in the "What changed in version x.y" should be enough, I'd think.  If their code breaks after upgrading versions, they'll look there and at least get a clue.

nophead, 2025-03-03:
It says deprecated because OpenSCAD used to automatically use a -1 step when the end was less than the start but now it doesn't. If it was before the last release I think everyone would have fixed their code by now.

andyDoucette, 2025-03-03:
Hey @nophead@nophead   Long time no see!  I remember all the respect I had for you from the RepRap days. :)  It's fun to see the entire wave of consumer level 3d printers come to be because of the RepRap project and everything that came from it.  It's lovely how a small open-source project like RepRap and openscad can change the world! :)

jordanbrown0, 2025-03-08:
@kintel , @t-paul : would you accept a PR to make "backwards" ranges yield an empty set, and drop the deprecation warning?

jordanbrown0, 2025-03-16:
Hmm.  These days it automatically reverses the two, so that `[10:0]` yields 0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10.  That's pretty unhelpful too.

With an explicit step and literal values for all three, you currently get a warning for a range where the start is "after" the end, whichever way that is.  Extending that warning to the implicit-step case seems reasonable, and would catch simple cases where people have messed up literal ranges.

commit 4b4d39e7 (2025-03-19): Ranges where start is "after" end (whichever way that is) yield nothing.  Fixes #5721. (#5750)

* Ranges where start is "after" end (whichever way that is) yield nothing.  Fixes #5721.

* Fix tests for "backwards" range semantics changes.
Review comment restoring comment.
closed 2025-03-19 by kintel