openscad/openscad #5722: Question: How to pass children to another module?

andyDoucette, 2025-03-02:
I'd like to pass some children from the top level, to a module and then from that module to another module.  How can I do that?

Example:
```
module inner_test()
{
    echo("inner_test() num children: ", $children);
}

module test()
{
    echo("test() num children: ", $children);

    inner_test()
        children();
}

test()
{
    cube(5);
    cube(6);
}
```

What I expect:
```
ECHO: "test() num children: ", 2
ECHO: "inner_test() num children: ", 2
```

What I get instead:
```
ECHO: "test() num children: ", 2
ECHO: "inner_test() num children: ", 1
```

This is probably me just not knowing how to use OpenSCAD right yet.  Please enlighten me.  I'm happy to learn! :)

UBaer21, 2025-03-02:
Does this help you?
```
module inner_test()
{
    echo("inner_test() num children: ", $children);
}

module test()
{
    echo("test() num children: ", $children);

    inner_test(){
        children(0);
        children(1);
    }
}

test()
{
    cube(5);
    cube(6);
}
```

jordanbrown0, 2025-03-02:
In the general case, where you don't know how many children there will be, alas, no, there is no way.

UBaer21, 2025-03-03:
@andyDoucette  your first module (test) sees 2 children (2×cubes) the second (inner_test) sees 1 children as you calling 1 × chidren() with "test".

Your first module can analyze the number and put into a variable so the second module get this data. Else it will count the number of "children()" modules from the module before.

```
module inner_test(children2)
{
    echo("inner_test() num children: ", children2);
}

module test()
{
    echo("test() num children: ", $children);

    inner_test(children2=$children)children();
}

test()
{
    cube(5);
    cube(6);
    cube(7);
}
```

in theory something like 
```
    inner_test(){
      if($children>0)children(0);
      if($children>1)children(1);
      if($children>2)children(2);
     }
```
could work, but this will always result in 3 children as the conditional is not evaluated for this, only the number of modules. So this can't be used.

kintel, 2025-03-03:
Yeah, I think we'd need some sort of spread/splat for this to work.

gsohler, 2025-03-03:
Andy , if you change to python mode, you can pass your "children" as
function arguments to any number of hierarchy levels though the functions.

You can even use python decorators to make your functions and even
primitives work "different"

nophead, 2025-03-03:
Perhaps this hacky version  will work
```
    if($children == 1) inner_test() { children(0); }
    else if($children == 2) inner_test() { children(0); children(1); }
    else if($children == 3) inner_test() { children(0); children(1); children(2); }
```

jordanbrown0, 2025-03-03:
In PR#4478 you can have arrays of geometry and arrays of module references, both of which might be useful.

andyDoucette, 2025-03-04:
Thanks but I'm wanting to use the children(vector) method.  I'd have to
enumerate every possible vector that could be passed, leading to factorial
complexity.

kintel, 2025-03-12:
Closing as this is more of a discussion on long-term potential features, which is best done on the mailing list.

closed 2025-03-12 by kintel

andyDoucette, 2025-03-12:
No because I want to ultimately use children(vector) in this use case, as
there is some logic that determines which children to pass to which inner
module.