openscad/openscad #5731: "Variables are set at compile-time, not run-time" is not really correct

jordanbrown0, 2025-03-08:
**Describe the bug**
The [manual](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/General#Variables_are_set_at_compile-time,_not_run-time) says "Variables are set at compile-time, not run-time".  That's not entirely accurate.  (It's not really accurate at all.)

OpenSCAD's parse time, the closest it comes to having a compile time, does not set variables.  It records the variable name and the expression assigned to it, but does not evaluate the expression.

At run time, it mostly straightforwardly executes assignments, evaluating expressions and assigning them to variables.  (Exception, not relevant here:  when you assign a variable twice the second assignment is executed as if it was at the location of the first.)

**To Reproduce**

```
module e(v) {
    echo(v=v);
}

e(1);
e(2);
```

It isn't even very meaningful to talk about v's value being set at compile time.

**Expected behavior**
OpenSCAD's variables are indeed ... interesting ..., but this is the wrong description.

jordanbrown0, 2025-03-08:
I may work on this later, but I noticed it in the middle of a different set of changes.

nophead, 2025-03-08:
`v` is a parameter, not a variable. Of course parameters can have different values on each invocation.

jordanbrown0, 2025-03-08:
```
module e(v) {
    v2 = v;
    echo(v2=v2);
}
```

nophead, 2025-03-08:
Local variables again are set per invocation and loop variables change per iteration. The manual is talking about them only having one value per scope. Global variables only have one scope and one value. Compile-time / run-time is not really relevant as you say because OpenSCAD doesn't really have a run time.

jordanbrown0, 2025-03-09:
Yes, of course they're set per invocation.  But the point of the issue is that, to the extent that OpenSCAD even has a "compile time", they're not set then.  (I'd say it definitely has a "run time".)

jordanbrown0, 2025-04-01:
I agree that it takes some mental gymnastics to get comfortable with OpenSCAD "variables".  But I just can't square "at compile time" with variables that are set based on a loop or based on arguments to a function or module.  A much closer analogy is that they are what C would call **const**, calculated whenever it's convenient (sometimes compile time, sometimes run time) but that cannot thereafter be changed.

bitbasher, 2025-04-05:
>... with variables that are set based on a loop or based on arguments to a function or module.
but but and but .. in a statement for( c = "test"){ // do stuff } the docs say explicitly that "c" is local to the scope created by the for statement. The docs say this:
   _for() is not an exception to the rule about variables having only one value within a scope.
   Each evaluation is given its own scope, allowing any variables to have unique values.
   No, you still can't do a=a+1;_ 
Which says to me that there will be 4 instances of the for statement, one for each letter in the "test" string. It is not a case of "c" being created in a symbol table and having 4 consecutive values as the loop runs .. it is that everything in the {} scope is "compiled", so to speak, into 4 execution modules in which all place holder names have their values calculated and embedded, which are then executed in order.

> ... A much closer analogy is that they are what C would call **const**, 
I don't know about recent versions of C, but in the bad old days of K&R C (the version i got deepest into ) a const was a truly immutable thing, created during compilation, its value calculated and stuffed into a space in the executable that could not be written to .. so i am not getting what you mean about a const being something that could ever be modified.

bitbasher, 2025-04-05:
not to be pedantic or anything :)

jordanbrown0, 2025-04-05:
Yes, it is entirely correct to think of "for" as creating N distinct scopes, newly creating variables on each iteration.

OpenSCAD is only weakly a compiled language.  The source is parsed and turned into a tree structure, that the execution engine then walks.  (Recursing into functions and modules as required.)  To the extent that there is any compilation phase, it is that parsing phase... which makes a single pass over the source, does no arithmetic, processes everything once regardless of "if" and "for", et cetera.  The execution engine does nothing that's at all like compilation.

K&R did not have **const**.  It had constants, which (roughly) were expressions that could be calculated entirely without doing anything other than arithmetic on constants.  ANSI C introduces the **const** modifier, which (in the part that's relevant here) creates a variable that can be set only via initialization, not via assignment.   A **const** (without **static**) inside a function is processed like any other automatic-variable initialization; its value can be based on variables, function calls, et cetera, and it is calculated each time the block is entered.  Thus it is entirely meaningful to say, for instance, `const time_t t = time(NULL);` and have `t` be the current time each time that block is entered, not allowed to be changed via a subsequent assignment.  (Note:  this is not a complete treatment of the behavior of **const**, but it's enough for this discussion.)

jordanbrown0, 2025-04-05:
I took a stab at rewriting this section, and did some nearby cleanup.  See the last few changes in the history.

closed 2025-04-05 by jordanbrown0