openscad/openscad #5736: Linux Dev Snapshot AppImage broken since 02/26/2025

keithel, 2025-03-11:
Tested under Ubuntu 24.04 x86_64, Downloading setting to executable and launching the Snapshot AppImage from February 26th, 2025 and beyond results in an error indicating that the SquashFS image uses (null) compression, and that this version only supports using xz, zlib, and ultimately execv fails (I assume because it could not mount the squashfs filesystem).

Testing any published AppImage snapshots from that date and beyond results in the same error.

**To Reproduce**
1. Go to https://files.openscad.org/snapshots/ - download any x86_64.AppImage published on or after 2025.02.26.
2. Set the .AppImage file to user executable (chmod u+x <AppImage file>)
3. Run the appImage file from a terminal.

The following console error output  will be emitted:
```
Squashfs image uses (null) compression, this version supports only xz, zlib.
ERROR: appimage_shall_not_be_integrated : sqfs_open_image error: /home/kyzik/Downloads/OpenSCAD-2025.02.26.ai23771-x86_64.AppImage
AppImageLauncher error: appimage_shall_not_be_integrated() failed (returned -1)
Squashfs image uses (null) compression, this version supports only xz, zlib.
ERROR: appimage_is_terminal_app : sqfs_open_image error: /home/kyzik/Downloads/OpenSCAD-2025.02.26.ai23771-x86_64.AppImage
AppImageLauncher error: appimage_is_terminal_app() failed (returned -1)
execv error: No such file or directory
```

**Expected behavior**
The expected behavior is that the AppImage launches, and you are greeted with OpenSCAD's opening window - which is a window allowing you to choose your poject to open or create a new project.

**Environment and Version info (please complete the following information):**
- OS: Ubuntu 24.04
- System: Intel x86_64
- OpenSCAD version - see above description.

t-paul, 2025-03-11:
Latest download works fine on Debian/Testing and https://github.com/TheAssassin/AppImageLauncher/issues/602 seems to indicate that this is an AppImageLauncher issue.

t-paul, 2025-04-27:
Closing as issue with AppImageLauncher not OpenSCAD.

closed 2025-04-27 by t-paul