openscad/openscad #5739: Some Characters in 3mf Metadata lead to faulty 3mf file

Nexusnui, 2025-03-13:
**Describe the bug**
3mf Metadata field allows some characters, leading to 3mf files not openable by some programs/libraries.

**To Reproduce**
1. Create and render model
2. Export -> 3mf
3. Check include metadata
4. Include " &" in Author field

**Expected behavior**
Either warning when using certain characters in the metadata or proper escaping of those characters.

**Code reproducing the issue**
Just any model

**Environment and Version info (please complete the following information):**
 - OS: Windows 11
 - System: PC 64-bit
 - OpenSCAD Version 2025.03.12 installed through Installer Exe

**Additional context**
I encountered the bug with Prusa Slicer and trimesh.

kintel, 2025-03-13:
It looks like we're doing it correctly. The metadata values can contain any string. We even escape special characters like `&`:
```
	<metadata name="Designer" preserve="1">&amp;Myself</metadata>
```

This feels like a bug in the software ingesting the 3MF files, not an OpenSCAD bug.

Nexusnui, 2025-03-13:
Weird that it happened in two different programs.

Nexusnui, 2025-03-13:
I think will report this to Prusa slicer and Trimesh, if this is not an issue with OpenScad.

kintel, 2025-03-13:
To double check, could you post your exported 3mf here, just so we can inspect the metadata entry?

Nexusnui, 2025-03-13:
This contains just cube 1x1x1 cube and "&" in the designer field.
GitHub did not allow me to upload a 3mf file so here is a google drive link -> https://drive.google.com/file/d/1se5qjBTtJP7PP5xVJ9oEDPY1QxTe8NOv

kintel, 2025-03-13:
Yours:
```
	<metadata name="Designer">Me & Other</metadata>
```

I guess you're using an OpenSCAD build with an older lib3mf version, where ampersands are not escaped.

@t-paul do you happen to have any insights here?

t-paul, 2025-03-13:
I'll have a look, but considering we use the official `lib3mf` interface, I would expect as application I can put in whatever I want and the library takes care of quoting. If not, I'd rate it a `lib3mf` bug - unless they document somewhere the app needs to take care of encoding, but that would be really strange.
It might also be different between v1 and v2 of the library. I think Windows still uses v1.

@Nexusnui If you want to attach any file, either just call it `file.3mf.txt` or put it into a ZIP archive. I believe both work.

Nexusnui, 2025-03-13:
I installed this version just with the installer from the OpenScad site, it is the build from yesterday.

Nexusnui, 2025-03-13:
I currently have two versions of OpenScad installed one under "C:\Program Files\OpenSCAD"(2024.04.22) and the current one under "C:\Program Files\OpenSCAD (Nightly)" (2025.03.12). Could the nightly  version pull wrong version of lib3mf?

t-paul, 2025-03-13:
No, I don't think there's anything on the installation side that can influence that. It's the build process that needs to be updated so future builds use the new library version.

Nexusnui, 2025-03-13:
It least there seems to be a (easy) solution.

t-paul, 2025-03-13:
Easy, well, sort-of. Very time consuming though. I'm building against both versions on Linux to see if the theory holds. I could not find a matching issue in the lib3mf github repo yet.

kintel, 2025-03-13:
According to the spec, the metadata content can be "any string" (xs:string), which makes literal `&` a legal character. I assume some apps wants to render these as HTML, and the lib3mf V2 perhaps decided to make life easier for them, even though the spec doesn't mandate it.

kintel, 2025-03-13:
For the record, my first metadata entry was generated on macOS using lib3mf-2.3.1

t-paul, 2025-03-13:
Merged into `lib3mf-v2-develop`: https://github.com/3MFConsortium/lib3mf/pull/143

t-paul, 2025-03-13:
[v1-1.8.1.3mf.zip](https://github.com/user-attachments/files/19237085/v1-1.8.1.3mf.zip)
```
<metadata name="Application">OpenSCAD (https://www.openscad.org/)</metadata>
<metadata name="CreationDate">2025-03-13T21:17:59Z</metadata>
<metadata name="Designer">lib3mf version: 1.8.1</metadata>
<metadata name="Description">Quoting & Fun - <br> 1 " 2 ' 3</metadata>
```

[v2-2.3.2.3mf.zip](https://github.com/user-attachments/files/19237086/v2-2.3.2.3mf.zip)
```
<metadata name="Application" preserve="1">OpenSCAD (https://www.openscad.org/)</metadata>
<metadata name="CreationDate" preserve="1">2025-03-13T21:29:43Z</metadata>
<metadata name="Designer" preserve="1">lib3mf version: 2.3.2</metadata>
<metadata name="Description" preserve="1">Quoting &amp; Fun - &lt;br&gt; 1 &quot; 2 &apos; 3</metadata>
```

jordanbrown0, 2025-03-13:
> I assume some apps wants to render these as HTML, [...]

No, this is an XML document and `&` is magic in XML and so must be escaped.  Similarly `<`.

kintel, 2025-03-13:
> > I assume some apps wants to render these as HTML, [...]
> 
> No, this is an XML document and `&` is magic in XML and so must be escaped. Similarly `<`.

I  thought the `&` escaping was an HTML thing, but I'm too tired to read the spec right now :)

jordanbrown0, 2025-03-13:
XML and HTML are cousins and share many rules.

It's not the specification, but here's a reference: https://en.wikipedia.org/wiki/XML#Escaping

t-paul, 2026-08-15:
Should be fixed via #6891.

closed 2026-08-15 by t-paul

Nexusnui, 2026-08-15:
After a quick test I can also say, that this bug is fixed.