openscad/openscad #5741: Manifold rendering issues with `rotate_extrude` on specific angles

vivienm, 2025-03-14:
**Describe the bug**

When combining a `rotate_extrude` with a simple, disjoint shape, the model may fail to build with the Manifold backend.

This depends on the angle given to `rotate_extrude`. On my experiments, this fails with values `135` and `225`, other values seem to work fine.

**To Reproduce**

Build the model (code below) with

```shell
openscad --hardwarnings --backend Manifold -o model.stl model.scad
```

This fails:

```
WARNING: PolySet -> Manifold conversion failed: NotManifold
Trying to repair and reconstruct mesh..
```

No problem with CGAL.

**Expected behavior**

The model should build successfully.

**Code reproducing the issue**

Here is the code:

```scad
// Angle matters. Seems to fail with 135 and 225, works with other values.
alpha = 135;

union() {
    cube(10);
    translate([30, 0]) rotate_extrude(angle=alpha) square(10);
}
```

**Environment and Version info (please complete the following information):**

- OS: Arch Linux
- System: Intel PC 64-bit
- OpenSCAD Version 2025.03.12 (AppImage)

**Library & Graphics card information**

<details>
```console
$ openscad --info
QObject::startTimer: Timers can only be used with threads started with QThread
OpenSCAD Version: 2025.03.12.ai24150 (git 7e8089da8)
System information: Linux 6.13.6-arch1-1 #1 SMP PREEMPT_DYNAMIC Fri, 07 Mar 2025 20:19:00 +0000 x86_64 Arch Linux 8 CPUs 31.06 GB RAM
User Agent: OpenSCAD/2025.03.12.ai24150 (git 7e8089da8) (Linux x86_64; Arch Linux)
Compiler: GCC "11.4.0" 64bit
MinGW build: No
Debug build: No
Boost version: 1_74
Eigen version: 3.4.0
CGAL version, kernels: 5.5.1, Cartesian<Gmpq>, Extended_cartesian<Gmpq>, Epeck
Python Version: Python 3.10.12
OpenCSG version: OpenCSG 1.8.1
Clipper2 version: 1.5.2
Manifold version: 3.0.1
Qt version: 5.15.3
QScintilla version: 2.11.6
InputDrivers:
GLib version: 2.72.4
lodepng version: 20230410
libzip version: 1.7.3
fontconfig version: 2.13.1 (runtime: 2.16.0)
freetype version: 2.11.1 (runtime: 2.13.3)
harfbuzz version: 2.7.4 (runtime: 10.4.0)
cairo version: 1.16.0
lib3mf version: 1.8.1
Features: roof, input-driver-dbus, lazy-union, vertex-object-renderers-indexing, textmetrics, import-function, predictible-output, python-engine
Application Path: /tmp/.mount_openscICgIlk/usr/bin
Documents Path: /home/vivien/.local/share
User Documents Path: /home/vivien
Resource Path: /tmp/.mount_openscICgIlk/usr/share/openscad
User Library Path: /home/vivien/.local/share/OpenSCAD/libraries
User Config Path: /home/vivien/.config/OpenSCAD
Backup Path: /home/vivien/.local/share/OpenSCAD/backups
OPENSCADPATH: /home/vivien/perso/projects/openscad/libraries
OpenSCAD library path:
  /home/vivien/perso/projects/openscad/libraries
  /home/vivien/.local/share/OpenSCAD/libraries
  /tmp/.mount_openscICgIlk/usr/share/openscad/libraries

OPENSCAD_FONT_PATH: <not set>
OpenSCAD font path:
  /home/vivien/.local/share/fonts
  /usr/local/share/fonts
  /usr/share/fonts
  /home/vivien/.nix-profile/share/fonts
  /nix/var/nix/profiles/default/share/fonts
  /home/vivien/.fonts
  /usr/X11/lib/X11/fonts
  /System/Library/Fonts
  /Library/Fonts
  /home/vivien/Library/Fonts
  /home/vivien/.local/share/fonts/caracteres
  /home/vivien/.local/share/fonts/marianne
  /home/vivien/.local/share/fonts/noto-emoji
  /usr/share/fonts/OTF
  /usr/share/fonts/TTF
  /usr/share/fonts/adobe-source-code-pro
  /usr/share/fonts/cantarell
  /usr/share/fonts/croscore
  /usr/share/fonts/encodings
  /usr/share/fonts/gsfonts
  /usr/share/fonts/liberation
  /usr/share/fonts/noto
  /usr/share/fonts/util
  /usr/share/fonts/encodings/large

GL context creator: EGL (new)
EGL version: 1.5
PNG generator: lodepng

GLAD version: 2.0.4
OpenGL Version: 4.6 (Compatibility Profile) Mesa 24.3.4-arch1.1
GL Renderer: Mesa Intel(R) Iris(R) Xe Graphics (TGL GT2)
GL Vendor: Intel
RGBA(8888), depth(24), stencil(8)
GL_ARB_framebuffer_object: yes
GL_EXT_framebuffer_object: yes
GL_EXT_packed_depth_stencil: yes
```

kintel, 2025-03-14:
This is a bit odd. It also looks like our tests failed to catch this as we don't have specific tests for that angle range.

I initially thought it was related to your object touching the y axis, but even adding a translation makes it fail.

kintel, 2025-03-14:
Minimal test case:
```
rotate_extrude(angle=135) translate([1,0]) square(10);
```

```
openscad issue5741.scad -o out.png --render=force --backend=manifold --hardwarnings                        
```

kintel, 2025-03-14:
Related to #5135

kintel, 2025-03-14:
Might be a duplicate of #5555

kintel, 2025-04-18:
Fixed by #5764