openscad/openscad #5749: Segv on closing edit tab

smartavionics, 2025-03-16:
**Describe the bug**
Using today's main branch, when I close the only edit tab, I get a SEGV, see backtrace below.

**To Reproduce**
1 - run openscad
2 - load a scad file (optional, it also crashes when you close the empty untitled.scad tab)
3 - click the close tab 

**Expected behavior**
No SEGV.

**Screenshots**
```
Thread 1 "openscad" received signal SIGSEGV, Segmentation fault.
0x000055555563f080 in QString::isEmpty() const ()
(gdb) where
#0  0x000055555563f080 in QString::isEmpty() const ()
#1  0x0000555556236e0d in MainWindow::checkAutoReload() ()
#2  0x00005555562653c8 in QtPrivate::FunctorCall<QtPrivate::IndexesList<>, QtPrivate::List<>, void, void (MainWindow::*)()>::call(void (MainWindow::*)(), MainWindow*, void**) ()
#3  0x000055555626338b in void QtPrivate::FunctionPointer<void (MainWindow::*)()>::call<QtPrivate::List<>, void>(void (MainWindow::*)(), MainWindow*, void**) ()
#4  0x000055555625fded in QtPrivate::QSlotObject<void (MainWindow::*)(), QtPrivate::List<>, void>::impl(int, QtPrivate::QSlotObjectBase*, QObject*, void**, bool*) ()
#5  0x00007ffff5385328 in QMetaObject::activate(QObject*, int, int, void**) () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#6  0x00007ffff53923ee in QTimer::timeout(QTimer::QPrivateSignal) () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#7  0x00007ffff5385bc5 in QObject::event(QEvent*) () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#8  0x00007ffff5d71a66 in QApplicationPrivate::notify_helper(QObject*, QEvent*) () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#9  0x00007ffff5d7b0f0 in QApplication::notify(QObject*, QEvent*) () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#10 0x000055555628775d in OpenSCADApp::notify(QObject*, QEvent*) ()
#11 0x00007ffff535980a in QCoreApplication::notifyInternal2(QObject*, QEvent*) () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#12 0x00007ffff53b0780 in QTimerInfoList::activateTimers() () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#13 0x00007ffff53b106c in  () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#14 0x00007ffff7bec17d in g_main_context_dispatch () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#15 0x00007ffff7bec400 in  () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#16 0x00007ffff7bec4a3 in g_main_context_iteration () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#17 0x00007ffff53b1435 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#18 0x00007ffff53583ab in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#19 0x00007ffff5360116 in QCoreApplication::exec() () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#20 0x00005555561f7b7b in gui(std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >&, std::filesystem::__cxx11::path const&, int, char**) ()
#21 0x0000555555b65b99 in main ()
```

**Environment and Version info (please complete the following information):**
```
System:    Host: toucan Kernel: 5.15.0-131-generic x86_64 bits: 64 Desktop: Xfce 4.16.1 
           Distro: Ubuntu 20.04.6 LTS (Focal Fossa) 
Machine:   Type: Laptop System: GIGABYTE product: GB-BSi3-6100 v: 1.x serial: <superuser/root required> 
           Mobo: GIGABYTE model: MKLP3AP-00 v: 1.x serial: <superuser/root required> UEFI [Legacy]: American Megatrends v: F5 
           date: 01/29/2016 
CPU:       Topology: Dual Core model: Intel Core i3-6100U bits: 64 type: MT MCP L2 cache: 3072 KiB 
           Speed: 889 MHz min/max: 400/2300 MHz Core speeds (MHz): 1: 889 2: 883 3: 894 4: 861 
Graphics:  Device-1: Intel Skylake GT2 [HD Graphics 520] driver: i915 v: kernel 
           Display: x11 server: X.Org 1.20.13 driver: modesetting unloaded: fbdev,vesa resolution: 1920x1080~60Hz 
           OpenGL: renderer: Mesa DRI Intel HD Graphics 520 (SKL GT2) v: 4.6 Mesa 21.2.6 
Audio:     Device-1: Intel Sunrise Point-LP HD Audio driver: snd_hda_intel 
           Device-2: Logitech HD Pro Webcam C920 type: USB driver: snd-usb-audio,uvcvideo 
           Sound Server: ALSA v: k5.15.0-131-generic 
Network:   Device-1: Intel Ethernet I219-V driver: e1000e 
           IF: enp0s31f6 state: up speed: 1000 Mbps duplex: full mac: 40:8d:5c:f4:c2:44 
           Device-2: Intel Wireless 3165 driver: iwlwifi 
           IF: wlp1s0 state: down mac: ac:2b:6e:15:50:54 
Drives:    Local Storage: total: 1.82 TiB used: 1.04 TiB (56.9%) 
           ID-1: /dev/sda vendor: Samsung model: SSD 870 EVO 1TB size: 931.51 GiB 
           ID-2: /dev/sdb type: USB vendor: Western Digital model: WD Elements 2621 size: 931.48 GiB 
Partition: ID-1: / size: 915.32 GiB used: 296.82 GiB (32.4%) fs: ext4 dev: /dev/sda5 
Sensors:   System Temperatures: cpu: 47.0 C mobo: 29.8 C 
           Fan Speeds (RPM): N/A 
Info:      Processes: 289 Uptime: 39d 21h 27m Memory: 15.49 GiB used: 7.73 GiB (49.9%) Shell: bash inxi: 3.0.38 
```

**Library & Graphics card information**
```
[... 409 lines cut ...]
GL_MESA_texture_signed_rgba
GL_MESA_window_pos
GL_NV_blend_square
GL_NV_compute_shader_derivatives
GL_NV_conditional_render
GL_NV_depth_clamp
GL_NV_fog_distance
GL_NV_fragment_shader_interlock
GL_NV_half_float
GL_NV_light_max_exponent
GL_NV_packed_depth_stencil
GL_NV_primitive_restart
GL_NV_texgen_reflection
GL_NV_texture_barrier
GL_NV_texture_env_combine4
GL_NV_texture_rectangle
GL_OES_EGL_image
GL_OES_read_format
GL_S3_s3tc
GL_SGIS_generate_mipmap
GL_SGIS_texture_border_clamp
GL_SGIS_texture_edge_clamp
GL_SGIS_texture_lod
GL_SUN_multi_draw_arrays
```

kintel, 2025-03-16:
Just to verify: Does the issue go away if you disable Design->Automatic Reload and Preview?

kintel, 2025-03-16:
This might be related to #5668 where we fixed some exit-related issues.

kintel, 2025-03-16:
Also, if someone could test Qt5 vs. Qt6 on Linux and report if this affects only one or both, that would be awesome.

smartavionics, 2025-03-16:
> Just to verify: Does the issue go away if you disable Design->Automatic Reload and Preview?

Yes, no segv when that is disabled.

kintel, 2025-03-17:
I can reproduce - will look into it.

kintel, 2025-03-17:
Hm, one is not supposed to be able to close the last editor, only hide it. @damienmarchal do you remember anything related to this?

damienmarchal, 2025-03-17:
Hi all,

Thank @smartavionics  for the bug report and @kintel  for pointing. 
This is definitively the kind of breaks that may have been introduced in the UX related PR I made recently. 
I can investigate that wednesday 19 and then report to you

damienmarchal, 2025-03-17:
For the moment I only remember when working my PR of having been puzzled by these lines 
```
void TabManager::closeCurrentTab()
{
  assert(tabWidget != nullptr);

  /* Close tab or close the current window if only one tab is open. */
  if (tabWidget->count() > 1) this->closeTabRequested(tabWidget->currentIndex());
  else par->close();
}
```

kintel, 2025-03-17:
> For the moment I only remember when working my PR of having been puzzled by these lines
> [...]
> and concluded that I needed to understand more before touching.

What we used to do in the past was: Once you close the second-to-last tab, leaving only one tab left, we removed the actual tab header, making it impossible to close the last tab. This way, there was no way of having dangling references to a deleted editor.

That behavior still makes sense: An OpenSCAD instance without an editor makes limited sense, and if that happens, we should probably close the entire MainWindow. The challenge is that closing the MainWindow as a reaction to the last editor closing, we still need robustly handle dangling editor pointers.

..so I currently see two ways of fixing this:
1. Make it impossible to close the last tab (remove the close icon, or remove the tab header)
2. On last editor close, automatically close the MainWindow (note: Not quit, as other MainWindow instances may be open). This probably requires us to support null editors in various places, and make sure to set that editor to null on last editor close.

damienmarchal, 2025-03-19:
Done using approach (1).

kintel, 2025-03-20:
Fixed by #5759

closed 2025-03-20 by kintel