openscad/openscad #5752: ERROR: The given mesh is not closed! Unable to convert to CGAL_Nef_Polyhedron due to Minkowski() and Difference()

chrisspen, 2025-03-17:
**Describe the bug**
If I attempt to render an STL from a difference of a cube with a Minkowski() sum, Openscad gives me the error:

ERROR: The given mesh is not closed! Unable to convert to CGAL_Nef_Polyhedron.

**To Reproduce**

Run this script:

  module make_peg_wrench(slot_width=14, slot_length=32, slot_depth=29, buffer=10, ball_buffer=5/2, arm_length=70, handle_diameter=300, handle_angle=45){
      
      difference(){
          minkowski(){
              
              union(){
                  translate([0,-(arm_length+buffer)/2,0]){
                  
                      cube([slot_length+buffer,arm_length+buffer,slot_width+buffer],center=true);
                      
                  }
                  
                  translate([0,50,0]){
                      difference(){
                          cylinder(d=handle_diameter, h=(slot_width+buffer), center=true, $fn=50);
                          cylinder(d=handle_diameter-(slot_width+buffer)*2, h=(slot_width+buffer)+1, center=true, $fn=50);
                          rotate([0,0,-handle_angle/2])
                          translate([-1000/2,0,0])
                          cube([1000,1000,1000],center=true);
                          rotate([0,0,handle_angle/2])
                          translate([1000/2,0,0])
                          cube([1000,1000,1000],center=true);
                      }
                      
                      for(i=[-1:2:1])
                      rotate([0,0,45/2*i])
                      translate([0,-handle_diameter/2+(slot_width+buffer)/2,0])
                      cylinder(d=(slot_width+buffer), h=(slot_width+buffer), center=true, $fn=50);
                      
                      difference(){
                          color("green")
                          translate([0,-120,0])
                          cube([(slot_length+buffer+50),40,slot_width+buffer], center=true);
                          
                          // round cutouts
                          color("blue")
                          for(i=[-1:2:1])
                          translate([i*(slot_length+buffer*.7),-handle_diameter/2+(slot_width+buffer)*2.05,0])
                          cylinder(d=(slot_width+buffer)*1.5, h=(slot_width+buffer)+1, center=true, $fn=50);
                      }
                      
                  }
              
              }
              
          
              translate([0,-(arm_length+buffer)/2,0])    
              translate([(slot_length+buffer)/2,(arm_length+buffer)/2,(slot_width+buffer)/2])
              sphere(r=ball_buffer, $fn=50);
          }
          
          // slot cutout
          color("red")
          translate([(slot_length+buffer)/2,-(slot_depth+buffer)/2,(slot_width+buffer)/2])
          cube([slot_length,slot_depth+buffer+9,slot_width],center=true);
      }
  }

  make_peg_wrench();

**Expected behavior**
[... 442 lines cut ...]
GL_NV_alpha_to_coverage_dither_control
GL_NV_blend_square
GL_NV_compute_shader_derivatives
GL_NV_conditional_render
GL_NV_copy_depth_to_color
GL_NV_copy_image
GL_NV_depth_clamp
GL_NV_fog_distance
GL_NV_fragment_shader_interlock
GL_NV_half_float
GL_NV_light_max_exponent
GL_NV_packed_depth_stencil
GL_NV_primitive_restart
GL_NV_texgen_reflection
GL_NV_texture_barrier
GL_NV_texture_env_combine4
GL_NV_texture_rectangle
GL_OES_EGL_image
GL_OES_read_format
GL_S3_s3tc
GL_SGIS_generate_mipmap
GL_SGIS_texture_border_clamp
GL_SGIS_texture_edge_clamp
GL_SGIS_texture_lod
GL_SUN_multi_draw_arrays

pca006132, 2025-03-17:
Please try the nightly version with the manifold backend.

kintel, 2025-03-17:
In the latest nightly, this also works fine with the CGAL backend, but manifold is recommended (renders in < 1 second).

closed 2025-03-17 by kintel