openscad/openscad #5774: Joystick Input is not working on nightly flatpak build

nomike, 2025-03-25:
**Describe the bug**
On the nightly flatpak build, joystick input doesn't seem to be registered.

**To Reproduce**
Steps to reproduce the behavior:
1. Install openscad-nightly using flatpak
2. Open it.
3. Goto Preferences -> Axis and enable Joystick
4. Open a design and try to navigate the view with a joypad attached to the computer

**Expected behavior**
Using a joypad should work.

**Code reproducing the issue**
n/a

**Screenshots**
n/a

**Environment and Version info (please complete the following information):**
 - OS: Ubuntu 24.10
 - System: Lenovo ThinkPad T14 Gen 1 (AMD) (AMD64 CPU)
 - OpenSCAD Version: 2025.03.18 (installed via flatpak)

**Library & Graphics card information**
n/a

**Additional context**
Works with a source-compiled 2025.02.24 build, so I guess it's an issue with flatpak not allowing the application to access the joystick device files.

t-paul, 2025-03-25:
Please create in the [flathub](https://github.com/flathub/org.openscad.OpenSCAD) repo. As that is not fully under our control, I can't transfer the issue unfortunately.

closed 2025-03-25 by t-paul

nomike, 2025-03-25:
https://github.com/flathub/org.openscad.OpenSCAD/issues/49