openscad/openscad #5775: Dark mode does not work on compiled version on Ubuntu 24.10 using Gnome desktop

nomike, 2025-03-25:
**Describe the bug**
I'm running  a source compiled version of OpenSCAD on an Ubuntu 24.10 Desktop machine under Gnome Desktop.
When I switch the system to dark mode and open OpenSCAD, everything is till light.

**To Reproduce**
Steps to reproduce the behavior:
1. Switch system to dark mode in system settings
2. Run OpenSCAD

**Expected behavior**
OpenSCAD should appear dark as well.

**Code reproducing the issue**
n/a

**Screenshots**

[image]

**Environment and Version info (please complete the following information):**
 - OS: Ubuntu 24.10 with Gnome Desktop
 - System: AMD64
 - OpenSCAD Version: 2025.02.24, compiled from source

**Library & Graphics card information**
n/a

**Additional context**
n/a

t-paul, 2025-03-25:
Works fine on Debian/Testing. Please check you have the packages installed that are listed here: https://hannes.hauswedell.net/post/2023/12/10/darkmode/

nomike, 2025-03-25:
Package `libadwaita` does not exist in Ubuntu.
I've installed the others mentioned on that page and changed the gconf settings as mentioned.

OpenSCAD is still not in dark mode.

nomike, 2025-03-25:
Is there anything else I can try to help you identify the root cause?

t-paul, 2025-03-25:
Not sure yet. That description made it work for me, I guess we have to dive deeper but I don't know yet where and how exactly. Did you compile with Qt5 or Qt6?

nomike, 2025-03-25:
```plaintext
$ openscad --info
QObject::startTimer: Timers can only be used with threads started with QThread
OpenSCAD Version: 2025.02.24
System information: Linux 6.11.0-19-generic #19-Ubuntu SMP PREEMPT_DYNAMIC Wed Feb 12 21:43:43 UTC 2025 x86_64 Ubuntu 24.10 12 CPUs 45.28 GB RAM
User Agent: OpenSCAD/2025.02.24 (Linux x86_64; Ubuntu 24.10)
Compiler: GCC "14.2.0" 64bit
MinGW build: No
Debug build: No
Boost version: 1_83
Eigen version: 3.4.0
CGAL version, kernels: 5.6.1, Cartesian<Gmpq>, Extended_cartesian<Gmpq>, Epeck
OpenCSG version: OpenCSG 1.5.0
Clipper2 version: 1.5.2
Manifold version: 3.0.1
Qt version: 5.15.15
QScintilla version: 2.14.1
InputDrivers: 
GLib version: 2.82.1
lodepng version: 20230410
libzip version: 1.7.3
fontconfig version: 2.15.0
freetype version: 2.13.3
harfbuzz version: 9.0.0
cairo version: 1.18.2
lib3mf version: (not enabled)
Features: roof, input-driver-dbus, lazy-union, vertex-object-renderers-indexing, textmetrics, import-function, predictible-output
Application Path: /usr/local/bin
Documents Path: /home/nomike/.local/share
User Documents Path: /home/nomike/Documents
Resource Path: /usr/local/share/openscad
User Library Path: /home/nomike/.local/share/OpenSCAD/libraries
User Config Path: /home/nomike/.config/OpenSCAD
Backup Path: /home/nomike/.local/share/OpenSCAD/backups
OPENSCADPATH: <not set>
OpenSCAD library path:
  /home/nomike/.local/share/OpenSCAD/libraries
  /usr/local/share/openscad/libraries

OPENSCAD_FONT_PATH: <not set>
OpenSCAD font path:
  /home/nomike/.local/share/fonts
  /home/nomike/.local/share/flatpak/exports/share/fonts
  /var/lib/flatpak/exports/share/fonts
  /usr/local/share/fonts
  /usr/share/fonts
  /var/lib/snapd/desktop/fonts
  /home/nomike/.fonts
  /home/nomike/.guix-home/profile/share/fonts
  /usr/share/texmf/fonts/opentype/public/lm
  /usr/share/texmf/fonts/opentype/public/lm-math
  /usr/share/texmf/fonts/opentype/public/tex-gyre-math
  /usr/share/texmf/fonts/opentype/public/tex-gyre
  /usr/X11/lib/X11/fonts
  /System/Library/Fonts
  /Library/Fonts
  /home/nomike/Library/Fonts
  /usr/share/fonts/X11
  /usr/share/fonts/cMap
  /usr/share/fonts/cmap
  /usr/share/fonts/fonts-go
  /usr/share/fonts/opentype
  /usr/share/fonts/truetype
  /usr/share/fonts/type1
  /usr/share/fonts/woff
  /usr/share/fonts/X11/Type1
  /usr/share/fonts/X11/encodings
  /usr/share/fonts/X11/misc
  /usr/share/fonts/X11/util
  /usr/share/fonts/cmap/adobe-cns1
[... 88 lines cut ...]
  /usr/share/fonts/truetype/samyak-fonts
  /usr/share/fonts/truetype/teluguvijayam
  /usr/share/fonts/truetype/tuffy
  /usr/share/fonts/truetype/ubuntu
  /usr/share/fonts/type1/texlive-fonts-recommended
  /usr/share/fonts/type1/urw-base35
  /usr/share/fonts/woff/ebgaramond
  /usr/share/fonts/X11/encodings/large
  /usr/share/fonts/opentype/roboto/slab
  /usr/share/fonts/truetype/roboto/unhinted
  /usr/share/fonts/truetype/roboto/unhinted/RobotoTTF

GL context creator: GLX (new)
GLX version: 1.4
PNG generator: lodepng

GLEW version: 2.2.0
OpenGL Version: 4.6 (Compatibility Profile) Mesa 24.2.8-1ubuntu1~24.10.1
GL Renderer: AMD Radeon Graphics (radeonsi, renoir, LLVM 19.1.1, DRM 3.59, 6.11.0-19-generic)
GL Vendor: AMD
RGBA(8888), depth(24), stencil(8)
GL_ARB_framebuffer_object: yes
GL_EXT_framebuffer_object: yes
GL_EXT_packed_depth_stencil: yes
```

pozix604, 2025-06-04:
`OpenSCAD-2025.06.03.ai25586-x86_64.AppImage` dark mode does not work on Debian 12 Gnome.

pozix604, 2025-06-04:
`gsettings set org.gnome.desktop.interface gtk-theme 'Adwaita-dark'` does not appear to make a difference for me, even after restart.

jordanbrown0, 2025-11-11:
This looks exactly like #4900.  (Which I would say is reporting an entirely valid bug, but proposing an incorrect solution.)