openscad/openscad #5776: Dark mode breaks icons on toolbar buttons and menus in OpenSCAD nightly in flatpak

nomike, 2025-03-25:
**Describe the bug**
When enabling dark mode in Gnome Desktop, the nightly version of OpenSCAD does not turn dark.
Additionally the icons on the buttons on the toolbar are no longer visible.

Icons for entries in the menu are also gone and are only revealed while a menu item is mouse-overed.

Also the welcome screen is not dark as well.

**To Reproduce**
Steps to reproduce the behavior:
1. Switch to dark mode in Gnome Desktop
2. Start OpenSCAD nightly via Flatpak

**Expected behavior**
Icons on buttons and menu items should still be visible when in Dark Mode.
The welcome dialog, the main window, preferences window and basically everything else should adhere to dark mode.

**Code reproducing the issue**
n/a

**Screenshots**

[image]

**Environment and Version info (please complete the following information):**
 - OS: Ubuntu 24.10
 - System: amd64
 - OpenSCAD Version: 2025.03.18 [flatpak]

**Library & Graphics card information**
n/a

**Additional context**
n/a

t-paul, 2025-03-25:
Please create in the [flathub](https://github.com/flathub/org.openscad.OpenSCAD) repo. As that is not fully under our control, I can't transfer the issue unfortunately.

closed 2025-03-25 by t-paul

nomike, 2025-03-25:
https://github.com/flathub/org.openscad.OpenSCAD/issues/48