openscad/openscad #5777: Pressing ctrl+shift+v while zooming fast breaks the view

nomike, 2025-03-25:
**Describe the bug**
When you zoom in or out fast (unlock the ratchet-mechanism of your mouse wheel and give it a spin) and press ctrl+shift+v the view gets broken.

This could manifest in the axis and scale marks being twice the size as they should be or being really tiny. Sometimes also the object rendering looks distorted and panning is backwards (panning left when it should pan right for example).

I even managed to have OpenSCAD become unresponsive a few times getting the "This application does not respond" message from Gnome.

Switching to a different file doesn´t fix it.

Resetting the view however, brings everything back to normal.

**To Reproduce**
Steps to reproduce the behavior:
1. Disengage the ratchet on your mouse's scroll wheel and give it a hard spin
2. Press ctrl+shift+v on the keyboard while the scroll wheel is still spinning.

**Expected behavior**
Pressing the view-all shortcut during scrolling should not cause any weird behavior.

**Code reproducing the issue**
n/a

**Screenshots**

[image]

[image]

[image]

**Environment and Version info (please complete the following information):**
 - OS: Ubuntu 24.10
 - System: amd64
 - OpenSCAD Version: Multiple 2025.03.18 [flatpak], 2025.02.24 [compiled from source]

**Library & Graphics card information**
n/a

**Additional context**
This might as well happen during regular scrolling, but disengaging the ratchet on your mouse is a sure-fire way to reproduce the issue.
Works for me every time.

kintel, 2025-03-25:
Shift-wheel changes the field-of-view angle, which can severely distort the view. We could put some limits on that to avoid getting lost.

rcolyer, 2025-03-25:
I'll be honest, nomike, I thought maybe you were crazy when I read this...  But this is a real bug.  It also does not require ctrl-shift-v for me, but just shift.  However, I want to emphasize a nuance of what's being described.  I'm using a touchpad rather than a mouse, and swooshing two fingers up rapidly on the touchpad, which emulates letting a scrollwheel fly inertially.  And this does not happen for me when pressing shift and then scrolling, but I do see a different decay of the view when scrolling fast and THEN pressing shift while it is already in motion.

[image]

cube([10,5,2]); cube([2,5,10]);

[image]

nomike, 2025-03-25:
I was also just able to create this:

[image]

rcolyer, 2025-03-25:
Many varieties of deformities are possible, it seems.

[image]

nomike, 2025-03-25:
@kintel And yes, shift-wheel changing the field-of-view angle explains the font on the axis labels enlarging and shrinking.
However, I do not think OpenSCAD should ever freeze during that.

So there's definitely something fishy going on here.

I'm afraid this will be tough to reproduce in code...

nomike, 2025-03-25:
Could this deformation somehow be related to this odd behavior happening during regular scrolling?

https://github.com/user-attachments/assets/e38a75e5-2c44-40cc-9fc1-be1269c79030

kintel, 2025-03-25:
The regular zooming behavior looks like an artifact caused by OpenCSG not playing well with the OpenGL front clipping plane. Not much we can do about that, but the latest OpenCSG may have added some new features related to that.

rcolyer, 2025-03-25:
Okay.  I understand what kintel meant there.  The nature of the bug has to do with shift-scroll changing the FOV, which does more obvious spatial distortions in perspective mode, but in orthogonal view doesn't do much on going forward until it causes some sort of wrap-around errors as the FOV gets far out of bounds of 360.  Scrolling backwards with it causes the inversions, scrolling forward with it very far causes wraps.  So really it just needs capped (if the feature even has a reason to exist).

kintel, 2025-03-25:
I haven't managed to reproduce a freeze. If anyone as hints towards when a freeze happens that would be interesting to explore. I'm also on macOS, so rendering is slightly different.

rcolyer, 2025-03-25:
I can't get a freeze, but zooming in far with shift, and out far without shift, can quickly drive position reported at the bottom of the screen to nan, and distance to inf.  Which is wild.  And maybe that causes a freeze on some display setups.

t-paul, 2025-03-25:
From the other issue, the hang probably happens with:
```
OpenCSG version: OpenCSG 1.5.0

GLEW version: 2.2.0
OpenGL Version: 4.6 (Compatibility Profile) Mesa 24.2.8-1ubuntu1~24.10.1
GL Renderer: AMD Radeon Graphics (radeonsi, renoir, LLVM 19.1.1, DRM 3.59, 6.11.0-19-generic)
GL Vendor: AMD
```

Maybe try with the Mesa software renderer using `LIBGL_ALWAYS_SOFTWARE=1 openscad`. For me that changes the OpenGL driver from:
```
GLAD version: 2.0.4
OpenGL Version: 4.6 (Compatibility Profile) Mesa 25.0.1-2
GL Renderer: Mesa Intel(R) Graphics (RPL-P)
GL Vendor: Intel
```

to:
```
GLAD version: 2.0.4
OpenGL Version: 4.5 (Compatibility Profile) Mesa 25.0.1-2
GL Renderer: llvmpipe (LLVM 19.1.7, 256 bits)
GL Vendor: Mesa
```

kintel, 2025-03-25:
Yeah, I would get off OpenCSG-1.5 and GLEW and onto OpenCSG-1.6 and GLAD. ..especially if the hang cannot be reproduced in F6 render mode.

t-paul, 2025-03-25:
@nomike for the time beeing, it's recommended to use `-DUSE_BUILTIN_OPENCSG=ON` that will use the updated/included OpenCSG library and also switch to GLAD instead of GLEW for OpenGL function handling.

> Could this deformation somehow be related to this odd behavior happening during regular zooming?

That looks like the normal "camera inside an object" case.

nomike, 2025-03-25:
Running OpenSCAD like this:

`flatpak run --user org.openscad.OpenSCAD -DUSE_BUILTIN_OPENCSG=ON`

Results in an error with F5 and F6 rendering:

> Ignoring unknown variable "ON".

Which is displayed in the Error-Log.

kintel, 2025-03-25:
`-DUSE_BUILTIN_OPENCSG=ON` is a build flag, not a run-time flag.

nomike, 2025-03-25:
Ah...yes. of course.  Mea culpa!
Thanks for pointing this out.

closed 2026-10-25 by github-actions[bot]