openscad/openscad #5780: Center text on top of the label

manosvoula, 2025-03-26:
I am trying to center the text but it goes at the bottom left corner below the sign, what do i do wrong? Can somone please correct the code?
// Dimensions of the label
width = 120;  // Width in mm
height = 50;  // Height in mm
thickness = 5;  // Thickness in mm
hole_diameter = 5;  // Diameter of holes
hole_offset = 10;  // Distance of holes from edges

// Texts for the three labels
texts = [
    ["Nerantzia", "Please, respect me!", "I grow slowly and need care.", "Do not break my branches."],
    ["Lemon Tree", "Give me time to grow!", "I need sun, water, and love to give you lemons.", "Do not cut my branches."],
    ["Fig Tree", "My branches are sensitive!", "I grow fast, but if I break, I struggle to recover.", "Please, do not pull or break them."]
];

// Function to create the label
module create_label(title, text_lines) {
    // Base of the label
    difference() {
        cube([width, height, thickness]);

        // Holes for tying at each corner
        translate([hole_offset, hole_offset, 0]) cylinder(h = thickness, d = hole_diameter);
        translate([width - hole_offset, hole_offset, 0]) cylinder(h = thickness, d = hole_diameter);
        translate([hole_offset, height - hole_offset, 0]) cylinder(h = thickness, d = hole_diameter);
        translate([width - hole_offset, height - hole_offset, 0]) cylinder(h = thickness, d = hole_diameter);
    }

    // 3D title text (with linear_extrude for 3D text)
    translate([width / 2 - (len(title) * 5), height - 15, thickness])  // Calculate title position
        linear_extrude(height = 1.5) text(title, size = 5, halign = "left");  // Left alignment

    // 3D body text (with linear_extrude for 3D text)
    multiline(text_lines, size = 4, lineheight = 1.2, vcenter = true, halign = "center", valign = "center", font = "Archivo Black");
}

// Function to create multi-line text
module multiline(text, lineheight=1.2, vcenter=false, size=10, halign=undef, valign=undef, spacing=1.0, font=undef, direction=undef, language=undef, script=undef){
    // Calculate the center for the text relative to the label
    translate([0, (vcenter==true?1:0) * -(len(text)-1) * size * lineheight / 2, 0]) {
        for(i = [0 : len(text)-1]) {
            translate([width / 2 - (len(text[i]) * size) / 2, (len(text)-1-i) * size * lineheight, 0]) {
                linear_extrude(height = 1.5) text(text=text[i], size=size, halign=halign, valign=valign, spacing=spacing, font=font, direction=direction, language=language, script=script);
            }
        }
    }
}

// Combine all the labels
for (i = [0 : len(texts) - 1]) {
    translate([0, i * (height + 10), 0])
        create_label(texts[i][0], [texts[i][1], texts[i][2], texts[i][3]]);
}

kintel, 2025-03-27:
Please use the mailing list or IRC for support: https://openscad.org/community.html

closed 2025-03-27 by kintel

jordanbrown0, 2025-03-27:
Horizontally... in your inner `translate()` in `multiline()`, you're double-accounting the need to center the text.  Just translate it to `width/2`, and let the `halign="center"` do its job.  (And don't try to figure out how wide the text is on your own!)

Vertically... in multiline(), your translates are more or less correct, and will center relative to the origin, but you never translate that origin up into the middle of the block.  You need a `translate()` around the whole thing, probably around the call to `multiline()`.

If you need more than that, yes, mailing list or IRC please.

jordanbrown0, 2025-03-27:
Oh, and in Z you never lift the text up to the surface of the blocks.  That translate I mentioned around multiline() should probably do that too.