openscad/openscad #5784: MouseSelector picking doesn't work for certain types of CSG trees

kintel, 2025-03-28:
MouseSelector picking is implemented by modifying preview rendering to render ID colors instead of regular surface colors. The IDs are used to map back to an ID associated with each CSG tree leaf node. By knowing this ID, we can walk the CSG tree and determine what code was responsible for generating the pixel we clicked on.

The following is only tested on macOS:

Picking seems to be very dependent on which object is rendered:
* For primitives, it seems to always work well.
* For simple differences, it never works: `difference() { sphere(10; cube(10); }`
* For slightly more complex models, like Examples/Basics/CSG.scad, it works reasonably well, but breaks completely from certain camera angles.

kintel, 2025-03-28:
@damienmarchal

kintel, 2025-03-28:
Next step: Write a debug renderer for ID colors to validate framebuffer contents, and test this across platforms to isolate OS- or system-specific issues.

damienmarchal, 2025-03-28:
This remind me this #5274, maybe it is connected.

kintel, 2025-03-30:
I'm starting to think that this happens because rendering is done outside of Qt's paintGL() call. This can behave differently across platforms, as some platforms have stricter context management than others.
Not sure how to verify and address yet, but I've tested that `MouseSelector::select()` seems to do the correct thing when run inside paintGL().

The other (smaller and likely benign) observation I made is that the size of the picking canvas is different than the window size, likely because the offscreen context is created with logical rather than physical coordinates (i.e. not taking scale factor for high-res displays into account). This shouldn't matter, especially since the ultimate goal is to render a tiny ROI around the mouse pointer anyway :)

kintel, 2025-03-31:
Found the issue: We didn't make the GL context current before running the mouse selection.

PR #5789 (2025-04-04): MouseSelection fix: Make context current, manage state
Fixes #5784
closed 2025-04-04 by kintel