openscad/openscad #5785: Ungrouped export

vortex64, 2025-03-28:
Presently I must export my .stl and process it in another application to "ungroup" when the model is in several sub parts.

For example, I may use a routine that does a dovetail cut of a model in order to allow it to be printed piecemeal and more easily on a smaller format printer.   Example here: https://www.thingiverse.com/thing:6303111

It would be nice if there was an "ungrouped" export in OpenSCAD that would automatically separate the sub models.  Then using a sequential file name like Modelname001.stl,  Modelname002.stl, etc...

Presently I export from OpenSCAD.  I then import into Raise3d IdeaMaker.  I then use this program to "ungroup" the sub models.  I then select each piece and export.  

Granted, with much additional work it is possible to provide conditional code paths in an OpenSCAD script to render each component separately, but this can be cumbersome for both the coder and ultimate user.

I've looked through the issues and I did not find any Feature Requests related but if this request has been raised previously I apologize in advance.

jordanbrown0, 2025-03-29:
I didn't "get" that the intent here was that the model be automatically split based on its geometry, which is very different from the explicit-split mechanism suggested in #1041.

vortex64, 2025-03-30:
I have had success using python scripts to post process OpenSCAD stl export and splitting the stl apart into separate pieces. 
using this...
https://github.com/fsantini/python-stlsplitter
...and numpy-stl to convert the stl export to ascii.

What would be nice is to see if this can be ported to C++ and run as an optional post process inside OpenSCAD when the user exports stl.   Perhaps add an additional export menu item - such as "Export as .STL (split)"

If there are many sub items in the current render then this could become a lengthy process.  A warning dialog could be presented to the user before the export split occurs that could inform the user as such and give them the option to abort before proceeding.

I will see if I can setup my OpenSCAD build environment and see if I can make progress with this.

gsohler, 2025-03-31:
> The only case where this would make sense would be cases where the model
> that is split is imported as a single object. I'm not sure that's something
> OpenSCAD should do.
>
> Right now there is not even any official way to handle the parts in the
> script other than having the split happening at export time. This makes it
> very specialized and limited in possible uses.
>
> So unless there's additional ideas/suggestions on how to make it more
> generally usable, I'd vote to close as *wont-do*.

When I understand correctly,

vortex64 wants to print STL files, which are way too big for his printer.
So his strategy is to split them into many acceptable-size sub-objects
at adequate places.
He does this by difference()  lots of cube([100,100,0.1])  to create many
objects from one.

This is the exact time when there are multiple objects in OpenSCAD. but he
can only save all of them at once.

https://github.com/pythonscad/pythonscad/blob/master/src/python/pyfunctions.cc#L2241-L2304
could be a reference implementation

t-paul, 2025-03-31:
If the separate parts are *already known* to OpenSCAD there is no need to have a slicer style "split mesh" operation when doing an export.

vortex64, 2025-03-31:
Correct, but if one arbitrarily splits a completed model up then it is needed.   I'm not talking out of the realm here.   Many other open sourced and free commercial programs (Cura, IdeaMaker, FreeCAD) have this "ungrouping" capability.   I think OpenSCAD should have this feature also.

t-paul, 2025-03-31:
No, it's not need. Example with https://github.com/hugokernel/OpenSCAD_Dovetail

```
...

// Now, cut !

// Extract the first part...
intersection() {
    amazing_design();
    cutter(position=[0, 0, 0], dimension=dimension, teeths=teeth, male=true);
}

// ... and the second part
intersection() {
    amazing_design();
    cutter(position=[0, 0, 0], dimension=dimension, teeths=teeth, male=false);
}
```

This is generates 2 distinct meshes *known* to OpenSCAD. (Mis-)using the "lazy-union" feature this will already export 2 distinct meshes to 3mf which can be imported and separately handled in for example PrusaSlicer. That is 1 export, 1 file, 2 meshes and no split operation whatsoever in any of the 2 programs involved.

[dovetail-demo.3mf.zip](https://github.com/user-attachments/files/19542855/dovetail-demo.3mf.zip)

[image]

[image]

It's also a much safer operation as it does not depend on the 2 objects having no overlap. This type of declaration can even export 2 entangled rings as it's simply 2 distinct meshes. I've never seen a split operation that is able to handle this case, some even fail if Z projection does no separate things.

What use case can not be handled this way?

jordanbrown0, 2025-04-01:
We really shouldn't recommend using lazy union for this use case.

t-paul, 2025-04-01:
Hence "(**Mis-**)using".

gsohler, 2025-04-01:
Not sure if you can cut everything as simple as rendering the two halves of
a dovetail cut. (or multiple dovetails)
For example imagine such a interleaved design like this, which is just too
big for my  printer

https://www.thingiverse.com/thing:69718260
(yes,you would need another connection concept like small cylinders e.g.)

Imagine you have an idea, how to have the part split,but  it will be a lot
of extra effort fiddling and capturing the intentional parts into the
dovetail given cube() area.
Its way easier to define the "cut areas" rather than the "piece areas"

OpenSCAD team has traditionally been extremely creative in proposing
alternative complex  approaches rather than addressing the issue directly.
IMHO most famous example is creating recursion to counteract
OpenSCAD's scopes.

But finally it appears to me the intent o/fun of the thread/mailing list is
to develop a huge discussion much more than anything else ;)

vortex64, 2025-04-01:
t-paul,  You state it is not a need but it's utility has already been established by the simple fact that other applications offer ungrouping of .stl models as I have previously described.

A case not covered by your solution is if one downloads an .stl from the wild that has several sub parts that has not been separated.

This is a workflow convenience.   It is not a dire need.  There are workarounds.  But that does not mean it is not worth pursuing.   

The other side of the coin is that I don't think it is very hard to implement this.   I'll take a shot at it.  I'll let you know how I make out.

jordanbrown0, 2025-04-01:
One could imagine an argument to `import()` that would select which of several objects to import.  But for STL there's no obvious way to identify particular parts; you could tell that there are two parts but not which one is number 1 and which one is number 2.

One torture test for your split logic is a hollow sphere, which should split as one object but in many implementations would split as two.

gsohler, 2025-04-01:
Yes, inner holes of solids is something which I have not yet considered yet.
My algorithm would need one additional phase to link a hole to its solid.
It's a quite cheap operation as only one vertex of each result has to be
compared against the others.
We could check , if the filename contains an "%d" and would use this as a
template to output the split objects

vortex64, 2025-04-01:
t-paul - I am simply trying to avoid having  to jump to an external application any time I want to break a model apart.

There are times when I'm building up a model and I'm not concerned about making the model parametric or releasing an OpenScad script for the model.   Nor am I interested in making OpenScad aware of the separate parts.   The model may be worked on is separate pieces in separate scripts.

Let's say something like a multimeter case,  I'll start designing the overall ergonomic shape of the case and then I'll split it with a plane to make two clamshell halves.    I'll export the model and ungroup it in Ideamaker.

I might then start a new script to work on just the top half of the case.   In that script I'll import the top half of the case and hollow the insides and add screw bosses make switch holes, etc..

Then do the same for the bottom half of the case.  Then on the bottom half of the case I might want to make a sliding dovetailed  9v battery door.  (Here the meaning of dovetailed is slightly different than what we have descibed previously).

So then I will make some plane subtraction cuts and I will export the model and ungroup it again in Ideamaker.  

Now I'll have the battery door separate and I'll import that into a new script  In that script I might add on the snap tab to it when you slide the battery door closed it snaps into position in the case and maybe some curved bosses to hold the battery in place.

...and so on.

I have done the above on many occasions.

You keep asking me to justify.  Like I said I don't feel that I have to.   Other applications like FreeCAD, Blender, Ideamaker,  etc... have already decided the utility of breaking .stl models apart and have implemented this feature.

Perhaps you should contact those developers and tell them they have wasted time in implementing this feature.  Tell them to remove it.

Now, that being said, I am aware that .3mf is the way forward as well as having OpenSCAD be aware of the distinct meshes.  But this is not always a mandatory approach.

I'm sorry, but no.  I did not download your dovetail example, so sorry for missing the clearance.  But I felt that you are directing the discussion to a "how to do a proper dovetail" and that is not the issue under discussion here.

In you prior message you seem to indicate that there were .stl files that were "two entangled rings" that you never seen a split operation would work on.   I'm curious about that.   Can you post a link to such an un-splitable .stl?

t-paul, 2025-04-01:
> Nor am I interested in making OpenScad aware of the separate parts. The model may be worked on is separate pieces in separate scripts.

Ok. That is exactly why it feels backwards to how OpenSCAD normally describes geometry and exports *exactly* that. I guess it would be an option to have it as export-only option first. At this point STL has no option dialog yet, so that would also be needed in addition (similar to pdf and 3mf) and it needs some decision how to handle the case where already multiple objects exist (even though lazy-union may be removed or changed in the future). In case the language ever gets the ability to do selection it could be promoted to be accessible from there.

> You keep asking me to justify. Like I said I don't feel that I have to.

*You* are asking people to spend *their* spare time on supporting *your* workflow. So yes, you have to spend a tiny bit of effort too to explain your idea and reasoning. In addition that also makes sure that if someone works on this, the result will actually match what is asked for.

vortex64, 2025-04-01:
I think separation on import would help drive "making OpenSCAD aware of individual parts", and more importantly, help drive "making OpenScad coders aware of deploying multiple parts" and drive more usage of the .3mf format.   

But, as you say,  it is something that can come later.   

Regarding export I was thinking for simplicity just adding another menu item to the list of export options "Export as STL (split)..."   If there are separable items in the current render it just takes the export filename and concatenates -01,-02, etc...

I think there should be a warning dialog before the split export proceeds, "Warning - If there are many sub models to split this could take some time.  Proceed? Y/N?"

FWIW - I made an openscad model of two chain linked torus's.  Ideamaker had no problem ungrouping the model.  The python script I mentioned previously also ungrouped the two torus's successfully - albeit very slowly.  This I attribute to having to work on an ascii formated .stl .   Implementation inside Openscad would be working more efficiently and directly on the mesh data.

jimb15122, 2025-04-17:
May I suggest using the otherwise useless (for stl export) COLOR tag as the basis for the split.  the resulting stl files would append the color (possibly in hex format) to the base filename.  UI wise, all that would be needed is a new "export stl by color" option.

t-paul, 2025-04-17:
@jimb15122 sorry, no. That is very much out of scope here. It's specifically about the case *without* declaration in the script itself.

jimb15122, 2025-04-17:
The "by color" export mode might be applicable to other file types.  The cli could provide an option to select a single color, or all colors.  If possible, the appended color name should match the actual text specified in the color tag.  eg"green", "#00FF00FF", etc.   some color name formats may need to be cleaned for use in file names, or just use hex in those cases.

jordanbrown0, 2025-04-17:
As Torsten says, out of scope here.  Please discuss in #1608.

vortex64, 2025-04-17:
Just an FYI,  I have the development environment for OpenSCAD under MSYS setup and running.   I am beginning work on .stl split export.   I can't predict a time when I will have a suitable result.  I need to study all the architecture OpenSCAD first.